openbmc/openbmc #2863: FVT: openpower-update-manager[1129]: No corresponding read-only volume found.

cdeadmin, 2018-02-01:


anoo1, 2018-02-01:
During a PNOR update, if the new image won't fit on flash, the code update code would remove an older image to make space. This is done via 2 independent services, which remove the r/w and r/o volumes for that pnor version. These services can get into each other and while the r/w service is querying the device directory, the r/o service may remove the volume information which causes the r/w service to fail.

This has only been seen once so very very small time window when this race condition can happen. The result is an orphan pnor volume that would be cleaned up on the next code update or BMC reboot.
There solution might be to merge the 2 delete services into 1 so that the operations are serialized.

There are 2 requests to remove:
```Jan 30 04:19:25 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-79c8edec...
Jan 30 04:19:25 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-79c8edec...
```
The ro volume is at volume 0 and gets removed by the ro service:
```Jan 30 04:19:27 witherspoon kernel: block ubiblock8_0: released
Jan 30 04:19:27 witherspoon kernel[971]: [  169.160000] block ubiblock8_0: released
```
The rw service is doing a query on the /sys/ directory but the information for the volume 0 was just deleted by the ro service:
```Jan 30 04:19:26 witherspoon obmc-flash-bios[1364]: libubi: error!: cannot read "/sys/class/ubi/ubi8_0/type"
Jan 30 04:19:26 witherspoon obmc-flash-bios[1364]: ubirmvol: error!: cannot find UBI volume "pnor-rw-79c8edec"```

rfrandse, 2018-08-01:
<https://gerrit.openbmc-project.xyz/11725> erase: Sleep between erasing the RO and RW vols
Closes  #2863 FVT: openpower-update-manager[1129]: No corresponding read-only volume found.

closed 2018-08-21 by geissonator