openbmc/openbmc #2867: [IPMI] DCMI request failed because: Unspecified error (ff)

swe12345, 2018-02-02:
ipmi dcmi set_asset_tag command fails
```
bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx dcmi set_asset_tag FP5lknvRI3bSekGi8jt87xm08IflHSeiZlzR8wCyQSXTJxhgVFNGvEUXahkWkdS

    DCMI request failed because: Unspecified error (ff)

Error setting asset tag!
 Set Asset Tag:
```
System Info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-40"
VERSION_ID="v2.1-40-g3dd3273"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-40"
BUILD_ID="v2.1"
root@witherspoon:~#
```
```
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.119
        op-build-v1.19-321-gb0b02c4-dirty
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9.8
        hostboot-7050d0a
        linux-4.13.16-openpower1-p759b60e
        petitboot-v1.6.3-p3987b5f
        machine-xml-592a6dd
        occ-39d5490
        hostboot-binaries-b29e165
        capp-ucode-p9-dd2-v2
        sbe-b6b7bb7
```

dkodihal, 2018-02-23:
It was actually working because of a persisted inventory. Reopening, I'm going to push a fix for this. The commit message will explain the problem.

rfrandse, 2018-02-23:
<https://gerrit.openbmc-project.xyz/9157> ipmi: fix broken asset tag function
Resolves  #2867 [IPMI] DCMI request failed because: Unspecified error (ff)

closed 2018-02-26 by geissonator