openbmc/openbmc #2868: NPM install: Compilation error

sivassrr, 2018-02-02:
Tried to run npm install and got following errors continuously
```npm ERR! tar.unpack untar error /home/sivas/.npm/lodash/4.17.4/package.tgz
npm ERR! tar.unpack untar error /home/sivas/.npm/caniuse-lite/1.0.30000792/package.tgz
```

sivassrr, 2018-02-02:
Above is on ubuntu VM from Mac based system. Here with attaching debug log as well. 
[npm-debug.log](https://github.com/openbmc/openbmc/files/1689805/npm-debug.log)

sivassrr, 2018-02-02:
snippet of debug log as follows:
```verbose cwd /home/sivas/newopenbmcgui/phosphor-webui
17 error Linux 4.4.0-31-generic
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v4.2.6
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error openbmc_browser_interface@0.0.1 start: `npm run server`
22 error Exit status 1
23 error Failed at the openbmc_browser_interface@0.0.1 start script 'npm run ser
ver'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the openbmc_browser_inter
face package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     npm run server
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs openbmc_browser_interface
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls openbmc_browser_interface
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

spinler, 2018-02-02:
Works fine for me.

gtmills, 2018-02-07:
Works for me as well. Your npm is pretty old. The latest stable version is 5.6.0.

sivassrr, 2018-02-08:
Tried it today and it works for me. as well. > openbmc_browser_interface@0.0.1 server /Users/sivas/openbmcgui8feb2018/phosphor-webui
> webpack-dev-server --history-api-fallback --inline --progress --devtool source-map

 10% [0] building modules 1/1 modules 0 active                                / Project is running at http://localhost:8080/
webpack output is served from /
Content not from webpack is served from ./src/public
404s will fallback to /index.html
[0] Child                                                                     a        170 modules
webpack: Compiled successfully.
----------> Just took the latest phosphor-webui

closed 2018-02-08 by sivassrr