openbmc/openbmc #2870: phosphor-host-ipmid needs to remove use of legacy inventory interface

geissonator, 2018-02-02:
I see in https://github.com/openbmc/phosphor-host-ipmid/blob/master/apphandler.cpp#L167 that the code us using the old inventory object to get the firmware version level of the BMC.  Unfortunately I'm not sure if we have a 1 for 1 replacement here. May need to create something else that works with both our UBI based systems and non-UBI systems.

This ties in with #1163 that I'm working on.

geissonator, 2018-02-02:
FYI @tomjoseph83 @nasamuffin <Vernon - can't find github id>

anoo1, 2018-02-05:
There's a xyz/openbmc_project/software/ object implemented on all systems regardless of their filesystem layout that provides the BMC version (from the VERSION_ID field on the /etc/os-release) that can be used to replace the call to query the old bmc inventory version field.

geissonator, 2018-02-05:
Not sure if you'd like a new issues but I also see some references to a TODO story that has been closed out that is removing some legacy code talking to the old settings interface.

chassishandler.cpp:
```
130//TODO : Can remove the below function as we have
131//       new functions which uses sdbusplus.
132//
133//       openbmc/openbmc#1489
134int dbus_get_property(const char *name, char **buf)
....
201//TODO : Can remove the below function as we have
202//       new functions which uses sdbusplus.
203//
204//       openbmc/openbmc#1489
205
206int dbus_set_property(const char * name, const char *value)
```

rfrandse, 2018-02-20:
<https://gerrit.openbmc-project.xyz/9074> DCMI: Remove use of legacy inventory interface
Resolves  #2870 phosphor-host-ipmid needs to remove use of legacy inventory interface

closed 2018-03-13 by geissonator