openbmc/openbmc #2871: OpenBMC GUI:  Uglifying breaks it

spinler, 2018-02-02:
The uglifier plugin (webpack.optimize.UglifyJsPlugin(),)  that runs during the build script as part of web-packing causes the following:

```
Uncaught Error: [$injector:modulerr] Failed to instantiate module app due to:
Error: [$injector:modulerr] Failed to instantiate module app.serverHealth due to:
Error: [$injector:unpr] Unknown provider: e
http://errors.angularjs.org/1.6.8/$injector/unpr?p0=e
    at http://localhost:8080/app.e5627e1c4a875c99587e.js:6:571
    at http://localhost:8080/app.e5627e1c4a875c99587e.js:6:20467
    at r (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:19376)
    at i (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:19681)
    at Object.a [as invoke] (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:19902)
    at r (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:18748)
    at http://localhost:8080/app.e5627e1c4a875c99587e.js:6:18887
    at o (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:1003)
    at p (http://localhost:8080/app.e5627e1c4a875c99587e.js:6:18649)
    at http://localhost:8080/app.e5627e1c4a875c99587e.js:6:18833
```

Searching around indicates that this is caused by the code being written in a way that is unfriendly to the uglifier, and so a change is needed.

The plan is to turn off uglification until a fix for this is found, so the UI can at least still be used, just with a slower load time.

edtanous, 2018-02-02:
Instead of taking the heavy handed approach of disabling uglifyjs, why not simply disable name mangling?

spinler, 2018-02-02:
@edtanous Thanks for the suggestion.  We'll do that instead.

rfrandse, 2018-08-16:
<https://gerrit.openbmc-project.xyz/12006> Turn uglification back on
Fixes github bug  #2871 OpenBMC GUI:  Uglifying breaks it

gtmills, 2018-08-20:
https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-webui/+/12006/ was merged. Closing.

closed 2018-08-20 by gtmills