openbmc/openbmc #2872: Improvements for porting application local erorrs

devenrao, 2018-02-05:
Following changes need to be done based on review comments for issue 2705
https://gerrit.openbmc-project.xyz/#/c/8383/

1) elog-gen.py should be placed in /usr/bin against the existing /usr/share folder
2) Running make cannot go off and copy files to random places on your system. that's why there are two steps - make, and make install.

amboar, 2018-02-05:
Can we bless elog-gen.py with a few sane defaults for generating the logging header? Like looking in some default locations for the yaml and template files? That would make it much easier to use by hand.

Also, has anyone thought about *not* generating a monolithic header? My thought would have been to instead generate a header corresponding to each yaml document. Applications would then include the specific error types they require, rather than dragging everyone down with all the types. In the interim we could still generate `elog-errors.h`, but just have it `#include` all the individual headers.

devenrao, 2019-07-04:
this issue is no longer valid as there is new logging infrastructure being developed to tackle the above issue by Matt Spinler

closed 2019-07-04 by devenrao