openbmc/openbmc #2875: GUI doesn't display what the actual error log messages are

gtmills, 2018-02-05:
GUI just says "xyz.openbmc_project.Logging.Entry.Level.Error" for the error log. It doesn't actually tell you what the actual error log. 
[image]

In this case the error log was xyz.openbmc_project.Software.Version.Error.ManifestFileFailure.
curl -c cjar -b cjar -k -H "Content-Type: application/json" https://${bmc}/xyz/openbmc_prject/logging/enumerate
{
  "data": {
    "/xyz/openbmc_project/logging/entry/2": {
      "AdditionalData": [
        "_PID=1240"
      ], 
      "Id": 2, 
      "Message": "xyz.openbmc_project.Software.Version.Error.ManifestFileFailure", 
      "Resolved": 0, 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error", 
      "Timestamp": 1517864369563, 
      "associations": []
    }, 
    "/xyz/openbmc_project/logging/internal/manager": {}
  }, 
  "message": "200 OK", 
  "status": "ok"
}

rfrandse, 2018-02-09:
<https://gerrit.openbmc-project.xyz/8916> Display the "type" of event log
Resolves  #2875 GUI doesn't display what the actual error type is

gtmills, 2018-02-19:
The "Error" in this screenshot is the "Severity". The "High" in this screenshot also comes from a mapping of the severity. https://github.com/openbmc/phosphor-webui/blob/master/app/common/services/constants.js#L70
So since  "Severity" is already displayed on the GUI, just change the error log name to the type.

closed 2018-02-23 by geissonator