openbmc/openbmc #2877: witherspoon-psu-monitor logs a lot to journal in QEMU

geissonator, 2018-02-05:
Not sure if there's something we could add to QEMU to make this service happy, or if maybe we can add something to witherspoon-psu-monitor to not retry so many times with the device is not present?

We get a ton of these to the journal
```
Feb 05 22:02:18 witherspoon witherspoon-psu-monitor[1169]: Failed to read sysfs file
Feb 05 22:02:18 witherspoon witherspoon-psu-monitor[1169]: Failed to read from the device. This can be used by any application that gets a failure reading from a device. It is up to the implementation on how to react to this error based on the use-case. Refer to the interface specification that references this for additional information.
Feb 05 22:02:19 witherspoon witherspoon-psu-monitor[1168]: Failed to read sysfs file
Feb 05 22:02:19 witherspoon witherspoon-psu-monitor[1168]: Failed to read from the device. This can be used by any application that gets a failure reading from a device. It is up to the implementation on how to react to this error based on the use-case. Refer to the interface specification that references this for additional information.
Feb 05 22:02:19 witherspoon witherspoon-psu-monitor[1169]: Failed to read sysfs file
Feb 05 22:02:19 witherspoon witherspoon-psu-monitor[1169]: Failed to read from the device. This can be used by any application that gets a failure reading from a device. It is up to the implementation on how to react to this error based on the use-case. Refer to the interface specification that references this for additional information.
```

geissonator, 2018-02-05:
FYI @bjwyman @amboar

amboar, 2018-02-05:
I have a very hackish pmbus subsystem cooked up for qemu. This would allow us to model things like the power supplies more easily, but getting that upstream is at the bottom of the pile in terms of things to get done.

bjwyman, 2019-03-29:
The power supply and many other drivers for Witherspoon are still on the list of things to do in the qemu wiki: https://github.com/openbmc/qemu/wiki
Devices
* ir35221 (hardware monitoring)
* dps310 (pressure, temp)
* max31785a (fan control)
* bmp280 (pressure, temp)
* ucd90160 (witherspoon power-supply)
* cffps1 (witherspoon power-supply)

bjwyman, 2019-03-29:
All of those would also be useful for Swift.

bjwyman, 2020-04-08:
Still on that TODO list: https://github.com/openbmc/qemu/wiki

closed 2020-11-09 by stale[bot]