openbmc/openbmc #2881: A simple http server that redirects to https

gtmills, 2018-02-06:
Can't use `<bmc>/web/` to get to the GUI, have to use `https://<bmc>/web/`.. A simple http server that redirects to https would fix this.

geissonator, 2018-04-03:
Some initial thoughts on this one:
- Figure out how to get our current https server, phosphor-gevent to redirect
- Do something like https://github.com/openbmc/rest-dbus and look at port 80 and just redirect to https
- other thoughts?

Would prefer a solution that doesn't introduce another python application to startup

amboar, 2018-04-04:
Hack up a socket-activated socat unit serving up a pre-canned HTTP redirect response? I imagine it would be pretty responsive.

geissonator, 2018-04-04:
Yeah, that's a good idea Andrew.  Another thought from Brad that fits with the direction we want to go anyway is to get nginx in now that we have latest yocto 2.4.  I'll look into reviving https://gerrit.openbmc-project.xyz/#/c/6969/.  It improves our overall https performance and has an easy way to redirect.  If it gets difficult though we may look into a shorter term solution.

rfrandse, 2018-04-10:
<https://gerrit.openbmc-project.xyz/10045> Allow web socket over nginx https connection
Resolves  #2881 A simple http server that redirects to https

closed 2018-05-04 by geissonator