openbmc/openbmc #2882: phosphor-gpio-monitor does not report useful error information

shenki, 2018-02-06:
My system had the following error in the journal:

```
Feb 06 03:17:00 ozrom1-bmc systemd[1]: phosphor-gpio-monitor@checkstop.service: Main process exited, code=dumped, status=6/ABRT
Feb 06 03:17:00 ozrom1-bmc systemd[1]: phosphor-gpio-monitor@checkstop.service: Unit entered failed state.
Feb 06 03:17:00 ozrom1-bmc systemd[1]: phosphor-gpio-monitor@checkstop.service: Failed with result 'core-dump'.
Feb 06 03:17:02 ozrom1-bmc systemd-coredump[1305]: Process 1296 (phosphor-gpio-m) of user 0 dumped core.
Feb 06 03:17:02 ozrom1-bmc phosphor-dump-manager[928]: Dump not captured due to a cap.
```
After cleaning out old logs, I got a core dump. It reported:
```
$ gdb-multiarch -core core.phosphor-gpio-m.0.034bd518fb3648dba4da3c1645038bb3.2172.1517887234000000000000 
Core was generated by `phosphor-gpio-monitor --path=/dev/input/by-path/platform-gpio-keys-event --key='.
```
However, running that command produced a different error:

```
root@ozrom1-bmc:~# phosphor-gpio-monitor --path=/dev/input/by-path/platform-gpio-keys-event --key=
Usage: phosphor-gpio-monitor [options]
Options:
  --help                  Print this menu
  --path=<path>           Path of input device. Ex: /dev/input/event2
  --key=<key>             Input GPIO key number
  --polarity=<polarity>   Asertion polarity to look for. This is 0 / 1 
  --target=<systemd unit> Systemd unit to be called on GPIO state change
ERROR: Key not specified.
```
This is because the core dump does not contain the full path. The archive produced by phosphor-dump-manager can tell me what the command line was:
```
    _CMDLINE=phosphor-gpio-monitor --path=/dev/input/by-path/platform-gpio-keys-event --key=74 --polarity=1 --target=obmc-host-crash@0.target
```
However, the journal file did not contain enough information to diagnose the error. It simply said:
```
$ cat journal-pid-2172.log  | grep MESSAGE
    MESSAGE=Failed to open device path
    MESSAGE=The operation failed internally.
    MESSAGE=terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
    MESSAGE=  what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
```
It would be preferable if there was an error in the journal that said something along the lines of:
```
phosphor-gpio-monitor: unable to open path: "/dev/input/by-path/platform-gpio-keys-event"
```
This would point me straight at the reason for the failure, and allow me to work out why that path was not able to be opened.

gtmills, 2018-02-06:
Error came from https://github.com/openbmc/phosphor-gpio-monitor/blob/master/evdev.cpp#L25

ojayanth, 2018-02-06:
@shenki Existing Core dump initiated BMC dump collects only the journal log associated PID  to the application which triggered the core.

closed 2020-06-12 by stale[bot]