openbmc/openbmc #2883: systemd-journald fails to send stream file descriptor, delays Witherspoon shutdown

amboar, 2018-02-06:
From a system built after openbmc/openbmc@9f22e7cc334b8ddd22c8ef8523b80288478f684b, I observed a significant delay at shutdown:

```
root@witherspoon:/tmp# reboot
[  OK  ] Removed slice system-phosphor\x2dmax31785\x2dmsl.slice.
[  OK  ] Stopped target Fan Control Ready.
[  OK  ] Stopped target Start Host0.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-phosphor\x2ddiscover\x2dsystem\x2dstate.slice.
[  OK  ] Removed slice system-dropbear.slice.
         Stopping Phosphor GPIO checkstop monitor...
[  OK  ] Removed slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Webserver (pre).
[  OK  ] Removed slice system-fsi\x2dbind.slice.
[  OK  ] Removed slice system-obmc\x2dsync\x2dinventory\x2ditem.slice.
[  OK  ] Removed slice system-obmc\x2dflash\x2dbmc\x2dsetenv.slice.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped PCIE Slot Presence Detection For Chassis0.
[  OK  ] Removed slice system-pcie\x2dslot\x2ddetect.slice.
         Stopping Phosphor Host Console SSH Per-Connection (9.81.199.114:47606)...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped target Chassis0 power on after reset.
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2don.slice.
         Stopping Phosphor poweron watchdog...
[  OK  ] Removed slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2drunning.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/network/config.
         Stopping Phosphor Fan Control Daemon...
[  OK  ] Removed slice system-phosphor\x2dfan\x2dmonitor\x2dinit.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-op\x2docc\x2denable.slice.
[  OK  ] Stopped target Timers.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Fan Monitor Daemon...
[  OK  ] Stopped target Host0 running after reset.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Removed slice system-mapper\x2dsubtree\x2dremove.slice.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Chassis0 (Power On).
[  OK  ] Stopped Witherspoon Fan Watchdog Reset.
[  OK  ] Stopped Wait for /xyz/openbmc_project/state/chassis0.
[ 3706.670000] watchdog: watchdog0: watchdog did not stop!
         Stopping Witherspoon Fan Watchdog Daemon...
         Stopping Phosphor Fan Presence T[ 3707.280000] occ-hwmon occ1-dev0: occ bus op failed rc:-4
ach Daemon...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-phosphor\x2dcooling\x2dtype.slice.
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped Wait for /xyz/openbmc_project/control/host0.
[  OK  ] Stopped Assert bmc_booted LED.
[  OK  ] Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
[  OK  ] Stopped target Host0 (Started).
[  OK  ] Stopped target Host0 (Starting).
[  OK  ] Stopped Start host0 SBE.
[  OK  ] Stopped target Start Host0 (Pre).
[  OK  ] Stopped Power on bind ir35221 device driver devs.
[  OK  ] Removed slice system-ir35221\x2don\x2dbind.slice.
[... 265 lines cut ...]
[ 3802.070000] systemd-journald[701]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
[ 3858.290000] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[ 3858.290000] systemd-shutdown[1]: Sending SIGKILL to PID 701 (systemd-journal).
[ 3858.470000] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[ 3858.470000] systemd-shutdown[1]: Unmounting file systems.
[ 3858.470000] systemd-shutdown[1]: Remounting '/var' read-only with options ''.
[ 3858.470000] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[ 3858.470000] systemd-shutdown[1]: Unmounting /var.
[ 3858.670000] UBIFS (ubi0:2): un-mount UBI device 0
[ 3858.670000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[ 3858.670000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[ 3858.670000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[ 3858.670000] systemd-shutdown[1]: All filesystems unmounted.
[ 3858.890000] reboot: Restarting system

U-Boot 2016.07 (Jan 31 2018 - 12:11:00 -0600)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:
```

geissonator, 2018-07-12:
looks like this one worked itself out:

```
         Starting Reboot...
[ 1684.974775] watchdog: watchdog0: watchdog did not stop!
[ 1685.010910] systemd-shutdow: 38 output lines suppressed due to ratelimiting
[ 1685.053900] systemd-shutdown[1]: Syncing filesystems and block devices.
[ 1685.215027] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[ 1685.250421] systemd-journald[1680]: Received SIGTERM from PID 1 (systemd-shutdow).
[ 1685.268173] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[ 1685.295149] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[ 1685.311465] systemd-shutdown[1]: Unmounting file systems.
[ 1685.327201] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" stops
[ 1685.342518] UBIFS (ubi8:3): un-mount UBI device 8
[ 1685.356874] UBIFS (ubi4:2): background thread "ubifs_bgt4_2" stops
[ 1685.381910] UBIFS (ubi4:2): un-mount UBI device 4
[ 1685.403276] UBIFS (ubi8:2): background thread "ubifs_bgt8_2" stops
[ 1685.418603] UBIFS (ubi8:2): un-mount UBI device 8
[ 1685.432787] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
[ 1685.448130] UBIFS (ubi8:1): un-mount UBI device 8
[ 1685.504566] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[ 1685.590815] UBIFS (ubi0:2): un-mount UBI device 0
[ 1685.599977] systemd-shutdown[1]: All filesystems unmounted.
[ 1685.605666] systemd-shutdown[1]: Deactivating swaps.
[ 1685.610866] systemd-shutdown[1]: All swaps deactivated.
[ 1685.616200] systemd-shutdown[1]: Detaching loop devices.
[ 1685.647535] systemd-shutdown[1]: All loop devices detached.
[ 1685.669517] reboot: Restarting system

U-Boot 2016.07 (Jul 12 2018 - 19:01:08 +0000)

```
v2.3-5-gaeb6a93

closed 2018-07-12 by geissonator

javashin, 2019-06-25:
hello im getting this problem

javashin, 2019-06-25:
im using systemd from git on gentoo and im getting this all the time