openbmc/openbmc #2885: Did SOL support input？

lockless, 2018-02-06:
BMC ： quanta-q71l  OpenBMC
X86-OS： Ubuntu-14.04

I use ipmitool to open SOL：
```
 ipmitool -I lanplus -H 192.168.23.226 -P 0penBmc sol activate
```
and i can see the x86-cpu OS (ubuntu) serial output through BMC IP-address, like that:
(when insert a USB storage device)
```
root@Ubuntu:/home/skunlin/onie/build/images# ipmitool -I lanplus -H 192.168.23.226 -P 0penBmc sol activate
Get Device ID command failed: 0xc1 Invalid command
[SOL Session operational.  Use ~? for help]
[ 1308.824522] sd 13:0:0:0: [sdb] No Caching mode page found
[ 1308.825661] sd 13:0:0:0: [sdb] Assuming drive cache: write through
[ 1308.827663] sd 13:0:0:0: [sdb] No Caching mode page found
[ 1308.828788] sd 13:0:0:0: [sdb] Assuming drive cache: write through
[ 1308.831327] sd 13:0:0:0: [sdb] No Caching mode page found
[ 1308.832536] sd 13:0:0:0: [sdb] Assuming drive cache: write through
```

the problem now is :  
how can i login the os on this SOL screen?
did OpenBMC support this?

thanks for advise!!
thanks for advise!!

chiabinglee, 2018-02-07:
As your information about the "Assuming drive cache: write through", it maybe the root cause that OS could not boot to login line.
[https://ubuntuforums.org/showthread.php?t=2306351](url)

bradbishop, 2018-04-12:
@pstrinkle @yuennancy fyi....

gtmills, 2018-08-20:
@lockless @bradbishop Still tracking something or can this be closed? Thanks!

closed 2018-08-24 by lockless