openbmc/openbmc #2887: [Witherspoon] HW CI : Power on failed to boot the Host

gkeishin, 2018-02-06:
Bunch of CI failure today . Looks like the communication with the host is broken somehow
So the host won't boot. 

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-60"
VERSION_ID="v2.1-60-g3426712"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-60"
BUILD_ID="v2.1"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-d7ccda2b787f8589c65f3943d854ee0ea9b69c50 #1 Thu Feb 1 05:07:40 UTC 2018 armv6l GNU/Linux
root@witherspoon:~#
```

```
Feb 06 16:38:24 witherspoon phosphor-chassis-state-manager[1281]: Received signal that power ON is complete
Feb 06 16:38:24 witherspoon phosphor-chassis-state-manager[1281]: Change to Chassis Power State
Feb 06 16:38:24 witherspoon openpower-proc-control[1769]: Setting SBE seeprom side to 0
Feb 06 16:38:24 witherspoon systemd[1]: Started Start host0 SBE.
Feb 06 16:38:24 witherspoon systemd[1]: Reached target Host0 (Starting).
Feb 06 16:38:24 witherspoon systemd[1]: Reached target Host0 (Started).
Feb 06 16:38:24 witherspoon systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
Feb 06 16:38:26 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/watchdog/host0.
Feb 06 16:38:26 witherspoon systemd[1]: Starting Start Watchdog0...
Feb 06 16:38:26 witherspoon phosphor-watchdog[1433]: watchdog: enabled and started
Feb 06 16:38:26 witherspoon systemd[1]: Started Start Watchdog0.
Feb 06 16:38:26 witherspoon systemd[1]: Reached target Start Host0.
Feb 06 16:38:27 witherspoon phosphor-host-state-manager[1335]: Received signal that host is running
Feb 06 16:38:27 witherspoon phosphor-host-state-manager[1335]: Change to Host State
Feb 06 16:38:27 witherspoon phosphor-timemanager[1336]: Changing time settings is *deferred* now
Feb 06 16:38:31 witherspoon ipmid[1306]: Host control timeout hit!
Feb 06 16:38:31 witherspoon ipmid[1306]: Timer expired
```

- [x] Journald 
[HW_CI_power_failure.txt](https://github.com/openbmc/openbmc/files/1700038/HW_CI_power_failure.txt)

geissonator, 2018-02-06:
This was a good one, on both machines, it was after testing https://gerrit.openbmc-project.xyz/#/c/6403/ that all jobs after it started failing.  A PDU cycle of both systems fixed the issue until we tested 6403 again, then it comes back.  Something is going on in 6403 that breaks things in a way that only a PDU cycle can fix it.

gkeishin, 2018-04-24:
@geissonator Do we still need this one.. or there is code mentioned there coming in again?

gtmills, 2018-10-12:
@geissonator @gkeishin  Can we close?

gkeishin, 2018-10-14:
Yeah.. I tink we can..

closed 2018-10-14 by gkeishin