openbmc/openbmc #2891: All DHCP parameters are getting set to default values after refreshObjects timer expires

ngorugan, 2018-02-08:
All DHCP config parameters are getting set to default values after refreshObjects timer expires.

ex:
// Reading DNSEnabled config parameter
root@witherspoon:~# busctl get-property xyz.openbmc_project.Network work/config/dhcp xyz.openbmc_project.Network.DHCPConfiguration DNSEnabled
b true
// Setting it to false
root@witherspoon:~# busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/config/dhcp xyz.openbmc_project.Network.DHCPConfiguration DNSEnabled b 0

// Reading  it to check if it is actually get set to false or not
root@witherspoon:~# busctl get-property xyz.openbmc_project.Network work/config/dhcp xyz.openbmc_project.Network.DHCPConfiguration DNSEnabled
b false

// Reading after sometime(after refreshObjects  timer expired)
root@witherspoon:~# busctl get-property xyz.openbmc_project.Network work/config/dhcp xyz.openbmc_project.Network.DHCPConfiguration DNSEnabled
b true

// Machine info
root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-58"
VERSION_ID="v2.1-58-g1052521"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-58"
BUILD_ID="v2.1"
root@witherspoon:~# cat  /proc/version
Linux version 4.10.17-d7ccda2b787f8589c65f3943d854ee0ea9b69c50 (anoo@gfw159.aus.stglabs.ibm.com) (gcc version 6.2.0 (GCC) ) #1 Mon Feb 5 13:37:06 CST 2018
root@witherspoon:~#

amboar, 2018-02-09:
Hello @ngorugan 

Please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

Cheers,

Andrew

rfrandse, 2018-03-07:
<https://gerrit.openbmc-project.xyz/9447> Get DHCP configuration values form n/w config file while creating DHCP config object.
Resolves  #2891 All DHCP parameters are getting set to default values after refreshObjects timer expires

closed 2018-03-14 by geissonator