openbmc/openbmc #2893: IPMI : Run op-test-framework

dkodihal, 2018-02-08:
Run IPMI related tests present in the op-test-framework against the OpenBMC IPMI stack. Fix reported problems.

ngorugan, 2018-03-07:
op-test-frame work has below given test suits for IPMI,
IPMI Test-suites:
1. BasicIPL -
Commands covered:
IPMI Power ON,
Power OFF,
Power Cycle
Power Soft,
Power Reset,
BMC Reset
BMC Reboot
IPMI Warm/Cold MC Reset
>Right now implemented using RestAPI for OpenBMC, PrudhviRaj is going to provide hack for IPMI.

2. full-outofbandipmi(pTestOOBIPMI)(lanplus interface)
Commands covered:
bmc, channel, chassis, dcmi, echo, event, exec, firewall, fru, lan, mc, pef, power, raw, sdr, sel, sensor, session, user
> Result:Total tests executed ="84", errors="14";skipped='24'
Test are actually executed twice(one time at Standby and one time at Runtime),but results are same in both cases.
Actual Total tests="42", errors="7";skipped='12'
```
1. error message="IPMI: command failed chassis poh" 
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc chassis poh; echo $?
Get Chassis Power-On-Hours failed: Invalid command
2.error message="Boot device is not set to floppy"
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc  chassis bootdev "floppy"
 >> This command is not working as expected.It is not setting boot device to floppy.
3. error message="IPMI: command failed dcmi get_conf_param"
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc dcmi get_conf_param; echo $?
    DCMI request failed because: Invalid command (c1)
Error Get DCMI Configuration Parameters!
4.error message="IPMI: command failed event 1"
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc event 1; echo $?
Platform Event Message command failed: Invalid command
5.error message="Failing to do hexdump for fru file" 
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc fru read 0 file_fru #no error, but not creating the file "file_fru"
6.error message="IPMI: command failed mc watchdog get"
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc mc watchdog get; echo $?
Get Watchdog Timer command failed: Invalid command
7.error message="IPMI: command failed -v raw 0x0 0xf"
ipmitool -H 9.3.185.83 -I lanplus -P 0penBmc -v raw 0x0 0xf; echo $?
Running Get PICMG Properties my_addr 0x20, transit 0, target 0x20
Invalid Get PICMG Properties response length 1
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0x20
Invalid response length 1
Discovered IPMB address 0x0
RAW REQ (channel=0x0 netfn=0x0 lun=0x0 cmd=0xf data_len=0)
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0xf rsp=0xc1): Invalid command
skipped:12
skipped message="OP AMI BMC specific" type="skip"
```

3. full-inbandipmi(OpTestInbandIPMI) (open interface)
OpTestInbandUsbInterface (usb or bt interface)
OpTestInbandIPMI – Host OS
SkirootFullInbandIPMI – Petitboot kernel
>Result:Total tests=“98”, errors=“16”;skipped=‘38’
few tests executed twice,
Actual Total tests=“65”, errors=“8”;skipped=‘32’
```
1.error message="Command 'ipmitool chassis poh' exited with 1.
2.error message="Expected to set floppy but instead got ['Boot parameter version: 1', 'Boot parameter 5 is valid/unlocked', 'Boot parameter data: 8000000000', ' Boot Flags :', '   - Boot Flag Valid', '   - Options apply to only next boot', '   - BIOS PC Compatible (legacy) boot ', '   - Boot Device Selector : No override', '   - Console Redirection control : System Default', '   - BIOS verbosity : Console redirection occurs per BIOS configuration setting (default)', '   - BIOS Mux Control Override : BIOS uses recommended setting of the mux at the end of POST']" type="AssertionError"
3.error message="Command 'ipmitool dcmi sensors' exited with -1.
>>Command 'ipmitool dcmi sensors' exited with -1
4.error message="Command 'ipmitool event 1' exited with 1
5.error message="Command 'hexdump -C /tmp/file_fru' exited with 1.
Output:
['hexdump: /tmp/file_fru: No such file or directory']" type="CommandFailed">
6.error message="Command 'ipmitool mc watchdog get' exited with 1.
Output:
['Get Watchdog Timer command failed: Invalid command']" type="CommandFailed">
7.error message="Command 'ipmitool pef info' exited with 1.
Output:
['IPMI command failed: Invalid command', 'Failed to get size of PEF Policy Table.', 'IPMI command failed: Invalid command', 'Failed to get PEF Capabilities.']" type="CommandFailed">
8.error message="Command 'ipmitool -v raw 0x0 0xf' exited with 1

skipped 3: 
with msg:test_sensor_get_host_status
skipped 32 :
with msg: OpenBMC doesn't support inband IPMI over USB
```
4. IPMI PNOR Reprovision(NVRAM, GARD)
>IPMI PNOR Reprovision are specific to AMI and SMC, not added raw commands for OpenBMC

5. IPMI Lock/Unlock Modes
>IPMI lock/unlock modes are specific to AMI and SMC, not added raw commands for OpenBMC

6. IpmiTortue test:
IPMI Torture test – standby
IPMI Torture test – runtime
IPMI SOL Console Torture test – standby
IPMI SOL Console Torture test – runtime
>Yet to run
7. BMC ResetTorture:
RuntimeBMCResetTorture - runtime
StandbyBMCResetTorture - standby
BMCResetvsHostIPLTorture – both
>Yet to run

ngorugan, 2018-03-07:
Tested using below given F/W version(GA2)
```root@witherspoon:~# cat /etc/os-release .
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.1"
VERSION_ID="ibm-v2.1-105-g4397421-r1-0-gbeb0b18"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.1"
BUILD_ID="ibm-v2.1-105-g4397421-r1"
```

ngorugan, 2018-03-20:
With latest GA2 build we are getting below given failures(it inlcudes both inband and outofband):
1.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc chassis poh;
```
Reason : we don't have support for this command(issue #2979 is open for it).
```
2.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc  chassis bootdev "floppy"
```
Reason: we don't have support to boot from  floppy.
```
3.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc chassis identify
```
Reason: support was there,but reverted because of a issue(#2740) on AST2400. It would be brought back once we move to yocto 2.4.
```
4.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc event 1
```
Reason:We don't have support for "Event Commands" for GA2
```
5.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc fru read 0 file_fru
```
Reason:For Device ID 0 it is failing because of "Device not present",for other devices it is failing because of space issue on BMC("No space left on device")
```
6.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc mc selftest

```
Reason: We don't have support(in code we are returning an error code:56H- Self Test function not implemented in this controller)
```
7.ipmitool -H <BMC_IP> -I lanplus -P 0penBmc -v raw 0x0 0xf;
ipmitool dcmi sensors
```
Reason:We don't have support. This is a raw command to Get POH Counter(issue#2979  should add support for this)
```
8.ipmitool dcmi sensors
```
Reason:TIMEOUT, This command roughly taking 2 min time. 
```
9.ipmitool pef info
```
Reason: We don't have support for "PEF and Alerting Commands" in GA2
```

With latest GA2 build  info:
```
root@witherspoon:/tmp# cat /etc/os-releaseID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.1"
VERSION_ID="ibm-v2.1-215-g6e7eacb-r3-cfm-0-gc07fe3d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.1"
BUILD_ID="ibm-v2.1-215-g6e7eacb-r3-cfm"
```

rfrandse, 2018-03-21:
<https://gerrit.openbmc-project.xyz/9647> Fixed broken "fru read" command
Resolves  #2893 IPMI : Run op-test-framework

closed 2018-03-27 by geissonator