openbmc/openbmc #2894: GUI: BMC IP should be BMC Host /IP in GUI header / server overview page

sivassrr, 2018-02-09:
Currently in GUI header, It marked as BMC IP. 

- [ ] It should be BMC Host / IP in gui header / server overview main page

[image]

rfrandse, 2018-02-16:
<https://gerrit.openbmc-project.xyz/9024> BMC IP changed to BMC host/IP in webui header/server overview page
Resolves  #2894 GUI: BMC IP should be BMC Host /IP in GUI header / server overview page

gtmills, 2018-02-21:
It also cuts off the .com 
[image]

jaypadath, 2018-02-22:
Hi Gunnar/Sivas, I don't the see the issue happening in recent witherspoon systems which I checked.
In place of IP address, only the IP address is seen.
Does it got fixed with some other change ?

jaypadath, 2018-03-02:
With further investigation it is observed that BMC host/IP address value
is shown depending on the user input. So kept this same design and made below
changes.

BMC IP changed to BMC host/IP in webui header/server overview page.
Also corrected the value truncation.
Attaching the fixed screen shot.
[image]

closed 2018-03-09 by geissonator