openbmc/openbmc #2897: FVT: Factory reset causes  obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service to crash

cdeadmin, 2018-02-09:


amboar, 2018-02-11:
@rfrandse @dkodihal @ratagupt can you please provide a description here outlining the issue, and detail what range of commits are affected?

rfrandse, 2018-02-14:
<https://gerrit.openbmc-project.xyz/8986> NetworkManager: Remove the create Dbus objects function
Resolves  #2897 FVT: Factory reset causes  obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service to crash

ratagupt, 2018-02-15:
This defect was to identify that why the < obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service> is failing during factory reset.

I would say not after factory reset it can occur any time but rare. 

It was failing that this service is having dependency on the network object,and mapper get the notification that the object has been created but when the application tries to access the d-bus object that object was not there.

closed 2018-02-20 by geissonator