openbmc/openbmc #2898: OP910:OPRAS:Witherspoon:GPUs not Listed as FRUs with openbmctool even after CUDA update

cdeadmin, 2018-02-09:


amboar, 2018-02-11:
@rfrandse @ratagupt can you please provide a description of the problem and the steps to reproduce, as well as the range of commits that are affected?

ratagupt, 2018-02-16:
Defect Symptom: 

OP910:OPRAS:Witherspoon:GPUs not Listed as FRUs with openbmctool even after CUDA update

On Witherspoon wsp004 with driver OP910 1742Fx and Pegas 1.0 [Red Hat Enterprise Linux Server 7.4 (Maipo) - Kernel 4.11.0-44.el7a.ppc64le on an ppc64le], after completing GPU Enablement Steps (CUDA update), when checking the FRU status of the GPUs, the "Is a FRU" values are set to no for all GPUs instead of Yes.:

./openbmctool.py -H wsp004 -U root -P 0penBmc fru status -v
Component    | Is a FRU  | Present  | Functional  | Assoc. Log Number(s)
gv100card0   | No           | Yes        | Yes            | None
gv100card1   | No           | Yes        | Yes            | None
gv100card3   | No           | Yes        | Yes            | None
gv100card4   | No           | Yes        | Yes            | None

Analysis:- 
openbmc tool script uses REST query  to get the above info.

openbmctool look at the FieldReplacable property to figure it out that whether the given inventory object is FRU or not.

Here GPU is not listed as FRU as we are creating the inventory object as part of set  sensor command and we don't implement the Replaceable interface for those inventory objects which is being created through set sensor.
Seems this is a new case which we need to fix.

Some technical insights
========================
gpu_func_sensor:
 serviceInterface: xyz.openbmc_project.Inventory.Manager
 readingType: assertion
 sensorNamePattern: nameLeaf
 interfaces:
    xyz.openbmc_project.State.Decorator.OperationalStatus:
      Functional:
        Offsets:
          0x08:
            type:   "bool"
            assert:  false
            deassert: true
        Prereqs:
          0x07:
            type: "bool"
            assert: true
            deassert: false
    xyz.openbmc_project.Inventory.Item:
      Present:
        Offsets:
          0x07:
            type: "bool"
            assert: true
            deassert: false

Here we can see that FieldReplacable interface is not there.

ratagupt, 2018-02-16:
After the fix the Rest inventory output looks like.
```    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card0": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card1": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card2": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card3": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card4": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/gv100card5": {
      "FieldReplaceable": 1, 
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
```

rfrandse, 2018-02-16:
<https://gerrit.openbmc-project.xyz/9026> IPMI Sensor: Adding the replacable interface for GPU sensor
Resolves  #2898 System:GPUs not Listed as FRUs with openbmctool even after CUDA update

closed 2018-02-21 by geissonator