openbmc/openbmc #2899: User power cap not making it to the OCC

cdeadmin, 2018-02-09:


amboar, 2018-02-11:
@rfrandse @dkodihal can you please provide a description, the steps to reproduce and the range of commits that are affected?

dkodihal, 2018-02-13:
Description: The power cap specified in a BMC setting is not making it to the host (OCC). The host code reads the power cap (which succeeds) and also a power supply derating factor (which is an IPMI sensor) for the cap computation. The derating factor that the host is reading is 0, which is incorrect.

This was most likely always broken.

Should be easily reproducible via an IPMI raw command. I'll post the command soon.

dkodihal, 2018-02-20:
I tried reading this sensor from the OS (that takes the host ipmi path), and could read back the expected value. The system had a build based off of https://github.com/openbmc/openbmc/commit/35bc9135afd0c1966ec752acf9a7b561ff387aaa.

ipmitool raw 0x04 0x2d 0xd8
0a 00 00 00

closed 2018-02-20 by dkodihal

spinler, 2018-02-20:
@dkodihal Isn't the derating factor supposed to be 90, and not 10?  https://github.com/openbmc/openpower-occ-control/blob/master/configure.ac#L130

mine260309, 2018-02-22:
@dkodihal @spinler I find this derating factor together with the Validator in settingsd confusing from end-user perspective.

The default validator ranges from 0 to 1000: https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-phosphor/settings/phosphor-settings-defaults/defaults.yaml#L44, but a system can override this value.

E.g. for Witherspoon:
1. The Validator is `500..3050`;
2. If a user wants to set power cap to 500, he is likely to set the property to 500;
3. Then settingsd checks the value and finds it valid, the property is changed to 500;
4. In openpower-occ-control, it multiplies with 0.9, so it will try to write 450 to occ;
5. However, OCC knows its valid value is 500~3050, so it produces an error.

Eventually, the power cap value is not written to OCC at all.
User will have to do calculation and write 556 to make it work.

What is the expected behavior on this?

spinler, 2018-02-22:
Hmm, those 500 and 3050 values are from the MRW, so they are output power values (derating factor already applied).  So, we should be showing to the customer a range of 556 to 3389.

mine260309, 2018-02-23:
OK, then it's a bug in phosphor-settings' yaml that the validator for power cap range shall be updated.

spinler, 2018-02-23:
@mine260309  Thanks for catching, that would have definitely annoyed people.