openbmc/openbmc #2901: netipmid: AssetTag property is not populated

gkeishin, 2018-02-09:
On latest master, looks like regression here. Easily re-creatable on latest Master

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-79"
VERSION_ID="v2.1-79-ga2efc8f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-79"
BUILD_ID="v2.1"
root@witherspoon:~#
```

`Journald Snippet`
```
Feb 09 07:02:56 witherspoon netipmid[1861]: AssetTag property is not populated
Feb 09 07:02:56 witherspoon netipmid[1861]: The operation failed internally.
```

`Test command`
```
ipmitool -I lanplus -C 3  -P 0penBmc -H xx.xx.xx.xx dcmi set_asset_tag YeWiK80W7R0EPlBeHwo0sOLWBuqr67nANerFUXVH0s2zFjnAzoaK4j5OVkhiFGJ 
```

gkeishin, 2018-02-09:
```
$ ipmitool -I lanplus -C 3  -P 0penBmc -H xx.xx.xx.xx dcmi set_asset_tag deadbeef

    DCMI request failed because: Unspecified error (ff)

Error setting asset tag!
 Set Asset Tag:
```

dkodihal, 2018-02-16:
This is the same as #2867.

closed 2018-02-16 by dkodihal