openbmc/openbmc #2902: Witherspoon : Inventory UUID object doesnt exist ?

gkeishin, 2018-02-09:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-79"
VERSION_ID="v2.1-79-ga2efc8f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-79"
BUILD_ID="v2.1"

/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
}
```
Object 

```
root@witherspoon:~# busctl introspect xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable interface -         -            -
.Introspect                         method    -         s            -
org.freedesktop.DBus.Peer           interface -         -            -
.GetMachineId                       method    -         s            -
.Ping                               method    -         -            -
org.freedesktop.DBus.Properties     interface -         -            -
.Get                                method    ss        v            -
.GetAll                             method    s         a{sv}        -
.Set                                method    ssv       -            -
.PropertiesChanged                  signal    sa{sv}as  -            -
root@witherspoon:~#
```

In working scenario, the field would show something like this
```
/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
    "BuildDate": "",
    "FieldReplaceable": 0,
    "Manufacturer": "IBM",
    "Model": "",
    "PartNumber": "01DH051",
    "Present": 1,
    "PrettyName": "BMC PLANAR  ",
    "SerialNumber": "000000000000",
    "UUID": ""
}
```

Looks like the object does not exist

```root@witherspoon:~# busctl get-property xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc xyz.openbmc_project.Common.UUID UUID
Unknown object '/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc'.
```

devenrao, 2018-02-09:
Tried the following commands on the failed system, just in case if it helps for debugging
```
root@witherspoon:~# busctl get-property xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc xyz.openbmc_project.Common.UUID UUID
Unknown object '/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc'.
```
```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP//xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/
{
  "data": [
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
  ],
  "message": "200 OK",
  "status": "ok"
}
```
```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP//xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
  },
  "message": "404 Not Found",
  "status": "error"
```
```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json"https://$BMC_IP//xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc/ethernet
{
  "data": {
    "FieldReplaceable": 0,
    "MACAddress": "",
    "Present": 1,
    "PrettyName": ""
  },
  "message": "200 OK",
  "status": "ok"
}
```
On another system 9.3.181.15 I noticed bmc is working but boxelder is not working
```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP//xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder"
  },
  "message": "404 Not Found",
  "status": "error"
}
```

```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP//xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
  "data": {
    "BuildDate": "",
    "FieldReplaceable": 0,
    "Manufacturer": "IBM",
    "Model": "",
    "PartNumber": "01DH051",
    "Present": 1,
    "PrettyName": "BMC PLANAR  ",
    "SerialNumber": "000000000000",
    "UUID": ""
  },
  "message": "200 OK",
  "status": "ok"
}
```

devenrao, 2018-02-09:
On failure system Bmc object details are not persisted
```
root@witherspoon:/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc# ls
ethernet
root@witherspoon:/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc#
```
On working system Bmc object details are persisted
```
root@witherspoon:/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc# ls -la
total 16
drwxr-xr-x    3 root     root           728 Jan  3 19:16 .
drwxr-xr-x    3 root     root           224 Jan  3 19:16 ..
drwxr-xr-x    2 root     root           584 Jan  3 19:16 ethernet
-rw-r--r--    1 root     root            81 Feb  8 11:59 xyz.openbmc_project.Common.UUID
-rw-r--r--    1 root     root           191 Feb  8 11:59 xyz.openbmc_project.Inventory.Decorator.Asset
-rw-r--r--    1 root     root            84 Feb  8 11:59 xyz.openbmc_project.Inventory.Decorator.Replaceable
-rw-r--r--    1 root     root           117 Feb  8 11:59 xyz.openbmc_project.Inventory.Item
-rw-r--r--    1 root     root             0 Feb  8 11:59 xyz.openbmc_project.Inventory.Item.Bmc
```

devenrao, 2018-02-09:
FFDC: http://tulpok01p4.aus.stglabs.ibm.com:8080/jenkins/job/OpenBMC/job/automated-test/job/helper-jobs/job/collect-bmc-ffdc/145/ If u need to look at it latter

amboar, 2018-03-05:
@devenrao Please do not link to IBM-internal resources. If necessary, bundle up that collection of files and attach it directly to the issue.

gkeishin, 2018-03-05:
Could this be due to VPD read related issue ?
```
root@witherspoon:~# systemctl status obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service | cat
● obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service - Sync system/chassis/motherboard/bmc/ethernet
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.wants/../obmc-sync-inventory-item@.service; disabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service.d
           └─mac_sync_inventory.conf
   Active: failed (Result: exit-code) since Mon 2018-03-05 08:40:13 UTC; 6min ago
  Process: 1262 ExecStart=/usr/bin/env sync_inventory_items.py -p ${PROPERTY} -s $SYNC_TYPE (code=exited, status=1/FAILURE)
 Main PID: 1262 (code=exited, status=1/FAILURE)

Mar 05 08:40:06 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/bmc/ethernet...
Mar 05 08:40:13 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 08:40:13 witherspoon systemd[1]: Failed to start Sync system/chassis/motherboard/bmc/ethernet.
Mar 05 08:40:13 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Unit entered failed state.
Mar 05 08:40:13 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Failed with result 'exit-code'.
```
```
root@witherspoon:~# journalctl --no-pager | grep -i pim
Mar 05 08:39:49 witherspoon openpower-read-vpd[1201]: PIM Notify() failed
root@witherspoon:~#
```

gkeishin, 2018-03-08:
Once we fixed the VPD I no longer see on the system.. So VPD for boxelder BMC corruption seems to be the culprit.

closed 2018-03-08 by gkeishin