openbmc/openbmc #2903: Witherspoon Missing BMC software association

gkeishin, 2018-02-09:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-79"
VERSION_ID="v2.1-79-ga2efc8f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-79"
BUILD_ID="v2.1"
root@witherspoon:~#
```

```
    "/xyz/openbmc_project/software/0999689b": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v2.1-79-ga2efc8f",
        "associations": [
            [
                "inventory",
                "activation",
                ""   <--- EMPTY associaition 
            ]
        ]
    },
    "/xyz/openbmc_project/software/0999689b/software_version": {
        "endpoints": [
            "/xyz/openbmc_project/software"
        ]
    },
```

- [x] Journald log : 
[BMC_association.txt](https://github.com/openbmc/openbmc/files/1710454/BMC_association.txt)

anoo1, 2018-03-02:
There's a VPD error on the system that's missing the association. The op-vpd-parser service creates the BMC inventory object so that may be the cause of the missing association.

```
Mar 02 17:57:00 witherspoon systemd[1]: Starting Read OpenPOWER-format VPD from EEPROM...
Mar 02 17:57:04 witherspoon openpower-read-vpd[1760]: PIM Notify() failed
Mar 02 17:57:04 witherspoon systemd[1]: Started Read OpenPOWER-format VPD from EEPROM.
Mar 02 17:57:04 witherspoon systemd[1]: Starting OpenBMC Software Update Manager...
```

gkeishin, 2018-03-08:
Once we fixed the VPD I no longer see on the system.. So VPD for boxelder BMC corruption seems to be the culprit.

closed 2018-03-08 by gkeishin