openbmc/openbmc #2905: Need to pass char arrays or char pointer to journal %s

geissonator, 2018-02-09:
I've seen this in a few different places in our code base.  We end up within things like this in the journal:

```
"MESSAGE" : "Untaring",
"FILENAME" : [ 136, 171, 232 ],
"EXTRACTIONDIR" : [ 168, 171, 232 ],
```

Which comes from this code:
```
228    log<level::INFO>("Untaring",
229                     entry("FILENAME=%s", tarFilePath),
230                     entry("EXTRACTIONDIR=%s", extractDirPath));
```

Where tarFilePath and extractDirPath are std::string

We need a .c_str() on the end of those for journal otherwise we get the weird numbers (I assume object address).  Need to do a grep through the repos with this issue and fix these up.

geissonator, 2018-02-12:
phosphor-host-ipmid/sensordatahandler.cpp#56 is another example:
```
54    if (mapperResponse.empty())
55    {
56        log<level::ERR>("Invalid mapper response",
57                        entry("PATH=%s", path),
58                        entry("INTERFACE=%s", interface));
59        elog<InternalFailure>();
60    }
```

anoo1, 2018-02-14:
The phosphor-bmc-code-mgmt repo is being fixed with this commit because it was needed to debug an issue, but there are other repos that remain to be fixed.
https://gerrit.openbmc-project.xyz/#/c/8980/

joseph-reynolds, 2018-05-03:
Would it be okay to specialize template function phosphor::logging::entry to handle std::string, or do you really want me to change all of its invocations?  I am thinking something like this:
```
template <typename Arg, typename ...Args>
constexpr auto entry<Arg, std::string>(Arg&& arg, std::string&& s)
{
    const auto entry_tuple = std::make_tuple(std::forward<Arg>(arg),
                                       s.c_str());
    return entry_tuple;
}
```

geissonator, 2018-05-03:
Seems reasonable to me, see any issues @dkodihal or @anoo1 ?

joseph-reynolds, 2018-05-10:
I've coded up the static_assert in a slightly different way (using template programming, not as a template specialization), asserting that the type of each parameter is either integral, floating, or pointer (e.g., char*) which corresponds to the types expected by the printf-formatters ("%c %d %f %g %s, etc").  It caught problems in 10 files, but I wish it was easier to read the compiler error messages.

I plan to fix this problem in two stages.  Stage 1: Add the string.c_str() as needed in various repos.  Then ensure the changes get through the code reviews and into the master level on github/openbmc before proceeding.  Stage 2: Add the static_assert to phosphor-logging.

joseph-reynolds, 2018-05-14:
I pushed changes for this in 6 separate gerrit reviews.  The phosphor-logging should be merged last or else it will cause static_asserts in other 5 repos.  See https://gerrit.openbmc-project.xyz/#/c/10604/

rfrandse, 2018-05-15:
<https://gerrit.openbmc-project.xyz/10604> static_assert typeof log message entries
Resolves  #2905 Need to pass char arrays or char pointer to journal %s

closed 2018-05-17 by geissonator