openbmc/openbmc #2907: nodejs failing to compile on ppc64le

geissonator, 2018-02-12:
Getting this in our docker containers running build-setup.sh on a ppc64le system
```
19:40:23 [161A[J[1;31mERROR[0m: [31mnodejs-native-4.6.1-r0 do_configure: Function failed: do_configure (log file is located at /tmp/openbmc/work/ppc64le-linux/nodejs-native/4.6.1-r0/temp/log.do_configure.22540)[0m
19:40:23 Currently  4 running tasks (4470 of 4653)  96% ||

19:40:23 0: systemd-1_232-r0 do_package_write_rpm - 42s (pid 21909)
19:40:23 1: systemd-1_232-r0 do_package_qa - 42s (pid 21936)
19:40:23 [161A[J[1;31mERROR[0m: [31mLogfile of failure stored in: /tmp/openbmc/work/ppc64le-linux/nodejs-native/4.6.1-r0/temp/log.do_configure.22540[0m
19:40:23 Log data follows:
19:40:23 | DEBUG: Executing python function sysroot_cleansstate
19:40:23 | DEBUG: Python function sysroot_cleansstate finished
19:40:23 | DEBUG: Executing shell function do_configure
19:40:23 | Usage: configure [options]
19:40:23 | 
19:40:23 | configure: error: option --dest-cpu: invalid choice: 'ppc64le' (choose from 'arm', 'arm64', 'ia32', 'mips', 'mipsel', 'ppc', 'ppc64', 'x32', 'x64', 'x86', 's390', 's390x')
19:40:23 | WARNING: exit code 2 from a shell command.
19:40:23 | ERROR: Function failed: do_configure (log file is located at /tmp/openbmc/work/ppc64le-linux/nodejs-native/4.6.1-r0/temp/log.do_configure.22540)
19:40:23 Currently  3 running tasks (4470 of 4653)  96% ||
```

amboar, 2018-02-12:
@gtmills it's not clear that your proposed change is correct as systems can have different endianness. My understanding is un-adorned powerpc{,64} builds are big-endian. We want little-endian support here.

rfrandse, 2018-02-15:
<https://gerrit.openbmc-project.xyz/8953> Add support for building on ppc64le
Resolves  #2907 nodejs failing to compile on ppc64le

gtmills, 2018-02-16:
Submitted a patch to oe, http://lists.openembedded.org/pipermail/openembedded-devel/2018-February/116737.html until OpenBMC moves to a version of yocto that includes the patch, we will need the bbappend.