openbmc/openbmc #2908: IPMID core dump on power on

gkeishin, 2018-02-12:
Seen on CI run.

```
Feb 12 02:56:46 witherspoon ipmid[1295]: Bus: :1.28, Path: /xyz/openbmc_project/sensors/chassis/PowerSupplyRedundancy, Interface: org.openbmc.SensorValue
Feb 12 02:56:46 witherspoon ipmid[1295]: *** Error in `ipmid': munmap_chunk(): invalid pointer: 0x00f65a00 ***
Feb 12 02:56:46 witherspoon ipmid[1295]: WATCHDOG RESET
Feb 12 02:56:46 witherspoon ipmid[1295]: WATCHDOG RESET
Feb 12 02:56:46 witherspoon ipmid[1295]: IPMI SET_SENSOR [0x03]
Feb 12 02:56:46 witherspoon ipmid[1295]: WATCHDOG RESET
Feb 12 02:56:46 witherspoon ipmid[1295]: IPMI GET_SENSOR_READING [0xd7]
Feb 12 02:56:46 witherspoon ipmid[1295]: IPMI GET_SENSOR_READING [0x0b]
Feb 12 02:56:46 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=6/ABRT
Feb 12 02:56:46 witherspoon phosphor-gevent[1182]: ::ca:6c76:8e91:f402%1995948604 - - [2018-02-12 02:56:46] "POST /login HTTP/1.1" 200 320 0.215689
Feb 12 02:56:46 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Feb 12 02:56:46 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Feb 12 02:56:47 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
Feb 12 02:56:47 witherspoon systemd[1]: Stopped Phosphor Inband IPMI.
Feb 12 02:56:47 witherspoon systemd[1]: Started Phosphor Inband IPMI
```

- [x] Journald 
[ipmid-coredump.txt](https://github.com/openbmc/openbmc/files/1715324/ipmid-coredump.txt)

geissonator, 2018-02-19:
This appears to have gone away, has not been seen in a while in CI.

gkeishin, 2018-02-19:
In Master also I don't see this one.. we can close this one for now.

closed 2018-02-19 by geissonator