openbmc/openbmc #2909: witherspoon: energy scale dcmi power set_limit range check invalid

cdeadmin, 2018-02-12:


spinler, 2018-02-12:
Premature bridging.  Closing.

gtmills, 2018-02-13:
For Witherspoon, the power cap range should be 500 to 3050 Watts, these values come from the MRW.
This is broke on master.

https://github.com/openbmc/openbmc/commit/a62ad5c1856b90752a49a0a0115399671806b83e broke this.

a62ad5c1856b90752a49a0a0115399671806b83e appended -native to native setting recipes to
follow bitbake naming conventions but left out this directory.
phosphor-settings-read-settings-mrw/ should have been renamed to phosphor-settings-read-settings-mrw-native/ since
phosphor-settings-read-settings-mrw.bb was renamed to phosphor-settings-read-settings-mrw-native.bb

closed 2018-02-15 by geissonator