openbmc/openbmc #2913: witherspoon phosphor-timemanager: Setting HostTime is not allowed

gkeishin, 2018-02-13:
Seeing this for some time on the latter Master builds

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-86"
VERSION_ID="v2.1-86-gb547eff"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-86"
BUILD_ID="v2.1"
root@witherspoon:~#

```

`Journald snippet`
```
Feb 13 06:59:15 witherspoon phosphor-timemanager[1303]: Setting HostTime is not allowed
Feb 13 07:10:23 witherspoon phosphor-timemanager[1303]: Setting HostTime is not allowed
```

`Time Manager Setting`
```
/xyz/openbmc_project/time/enumerate
{
    "/xyz/openbmc_project/time/bmc": {
        "Elapsed": 1518506401457931
    },
    "/xyz/openbmc_project/time/host": {
        "Elapsed": 1518506401475994
    },
    "/xyz/openbmc_project/time/owner": {
        "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.BMC"
    },
    "/xyz/openbmc_project/time/sync_method": {
        "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.NTP"
    }
}
```

geissonator, 2018-02-13:
Is there an ipmid trace near those?  Wondering if the host is trying to set the time, which would mean those logs are valid because the BMC owns the time.

gkeishin, 2018-02-13:
I don't see any obvious traces around it,  any Master or recent CI power on should have those traces. I asked around but couldn't get help. Will probably need @mine260309 to look at it.

gtmills, 2018-10-12:
@mine260309 @gkeishin Still a problem?

closed 2018-10-15 by gkeishin