openbmc/openbmc #2917: host checkstop monitoring appears to be broken

oohal, 2018-02-15:
When triggering a host checkstop the BMC side checkstop detection appears to be broken on witherspoon and possibly on other platforms. To reproduce, inject any host checkstop with pdbg from the BMC or with putscom on the host:
```
bmc # pdbg putscom -p0 0x02010800 0x4000000000000000

host # putscom -c0 0x02010800 0x4000000000000000
```

After injecting the host xstop the host will remain up (in the xstopped state) and we get this in the BMC journal:
```
root@w3:~# journalctl -f
-- Logs begin at Thu 2018-02-15 03:25:29 UTC. --
Feb 15 04:33:50 w3 kernel: ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
Feb 15 04:33:50 w3 kernel: ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
Feb 15 04:33:50 w3 kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI: HNCDSC AEN - channel 0 state down
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI: suspending channel 0
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI: configuring channel 0
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI: channel 0 config done
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI: No more channels to process
Feb 15 04:33:50 w3 kernel[1096]: [81242.950000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 15 04:35:05 w3 ipmid[1481]: Pushing cmd on to queue
Feb 15 04:35:05 w3 ipmid[1481]: Pushing cmd on to queue
Feb 15 04:35:05 w3 ipmid[1481]: Asserting SMS Attention
Feb 15 04:35:05 w3 ipmid[1481]: SMS Attention asserted
Feb 15 04:35:05 w3 systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-hub\x403400-cfam\x401\x2c0-sbefifo\x402400-occ\x402-occ\x2dhwmon\x402.service: Unit not needed anymore. Stopping.
Feb 15 04:35:05 w3 systemd[1]: Stopping Phosphor Hwmon Poller...
Feb 15 04:35:05 w3 systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 15 04:35:10 w3 ipmid[1481]: Pushing cmd on to queue
Feb 15 04:35:10 w3 ipmid[1481]: Pushing cmd on to queue
Feb 15 04:35:10 w3 ipmid[1481]: Command in process, no attention
Feb 15 04:35:10 w3 systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Feb 15 04:35:10 w3 systemd[1]: Stopping Phosphor Hwmon Poller...
Feb 15 04:35:10 w3 systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 15 04:35:35 w3 ipmid[1481]: Host control timeout hit!
Feb 15 04:35:35 w3 ipmid[1481]: Timer expired
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Host control signal values
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Error resetting the OCC.
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Host control signal values
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Error resetting the OCC.
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Host control signal values
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Error resetting the OCC.
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Host control signal values
Feb 15 04:35:35 w3 openpower-occ-control[1292]: Error resetting the OCC.
```

Previously I would get something like this immediately after a host checkstop:
```
Aug 03 16:01:25 witherspoon checkstop_app[2508]: Host checkstop condition detected
Aug 03 16:01:26 witherspoon phosphor-fru-fault-monitor[1168]: /xyz/openbmc_project/logging/entry/3
Aug 03 16:01:26 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Aug 03 16:01:26 witherspoon systemd[1]: Reached target Quiesce Target.
Aug 03 16:01:26 witherspoon systemd[1]: Reached target Host instance 0 crashed.
```
Which would be followed by the Host being either rebooted by the BMC or left in the Quiesce state for debug, depending on the BMC configuration. This behaviour should be restored.

OpenBMC version:
```
root@w3:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.1"
VERSION_ID="ibm-v2.1-55-g35bc913-r1-ctm-0-g8b7a6f2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.1"
BUILD_ID="ibm-v2.1-55-g35bc913-r1-ctm"
```
I've found this reproduces on any recent OpenBMC build though.

Host firmware:
```
{
  "data": {
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
    "ExtendedVersion": "buildroot-2017.11-5-g65679be,skiboot-v5.10-rc2,hostboot-de81205,linux-4.14.16-openpower1-pfefc0b0,petitboot-v1.6.6-pecf56cd,machine-xml-58554bf,occ-f72f857,hostboot-binaries-134d61b,capp-ucode-p9-dd2-v3,sbe-1f773f1",
    "Path": "/tmp/images/c2114377",
    "Priority": 0,
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active",
    "Version": "open-power-witherspoon-59789bf",
    "associations": [
      [
        "inventory",
        "activation",
        "/xyz/openbmc_project/inventory/system/chassis"
      ]
    ]
  },
  "message": "200 OK",
  "status": "ok"
}
```

spinler, 2018-02-15:
@oohal Could you humor me and read the status register that says the processor is checkstopped?  I would ask you to read the JTAG_ATTENTION GPIO but the resource would be in use by the checkstop monitor.

SrideviRamesh, 2018-02-15:
I see that error injection which u are using is not correct.
From the info .. Your injecting a CXAFIR checkstop UE through BMC & HOST.
pdbg -d p9w -p0 putscom 0x02010800 0x4000000000000000 ---> through BMC
putscom -c 0x0 0x02010800 0x4000000000000000  ----> through HOST

can u try this procedure on BMC & HOST.

when tried on BMC host should reach quiesced state.
when tried on HOST.. HOST will take a reboot & gard the resource & boots successfully to runtime.

oohal, 2018-02-20:
@SrideviRamesh I don't see what the difference between what you've quoted there and what I put in the original post. I verified that a checkstop was occurring with that injection before I filed the bug.

@spinler Sure, you can verify that a host checkstop is happening  by looking at the Global FIR summary register:
```
root@w3:~# pdbg getscom 0x570F001C -p0
p0:0x570f001c = 0x2000000000000000
```
PPC_BIT(2) being set indicates chiplet 2 has checkstopped. That chiplet corresponds to N0 which contains the CAPP unit that I just broke with the injection. Running fircheck from a cronus setup yields the same information in a slightly more readable format:
```
CPU Summary:
== Checkstops ==
     TARGET NAME | CHIPLET NAME | ADDITIONAL NAME |      ADDR (BIT) | DESCRIPTION
p9n:k0:n0:s0:p00 |           N0 |              -- | 0x02010800 (01) | CXA System Xstop PE 
```

Even after doing the injection the BMC still thinks the host is up and running:
```
root@w3:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Running
```

As for the GPIO, did you mean CHECKSTOP rather than JTAG_ATTENTION? Looks like the attention GPIO doesn't exist on witherspoon:
```
root@w3:~# gpioutil -n JTAG_ATTENTION
ERROR - GPIO Name doesn't exist
Valid names:
        BMC_CP0_PERST_ENABLE_R
        IDBTN
        CRONUS_SEL
        BMC_POWER_UP
        FSI_CLK
        BMC_CP0_RESET_N
        RESET_BUTTON
        FSI_DATA
        POWER_BUTTON
        CHECKSTOP
        SOFTWARE_PGOOD
        BMC_VS1_PERST_N
        FSI_ENABLE
        SYS_PWROK_BUFF
root@w3:~# gpioutil -n CHECKSTOP
GPIO ID: 354; DIRECTION: falling
[Errno 16] Device or resource busy
root@w3:~#
```
And yeah, looks like it's tied up already.

spinler, 2018-02-20:
Looks like pdbg is turning off the bits that enable the checkstop propagation to the FSI slave from the FSI2PIB engine:
```
# pdbg -b fsi -d p9w -p0 getcfam 0x100d
p0:0x100d = 0x60000000

(just writing 0s)
# pdbg putscom -p0 0x02010800 0x0000000000000000

# pdbg -b fsi -d p9w -p0 getcfam 0x100d
p0:0x100d = 0x00000000

spinler, 2018-02-20:
@cyrilbur-ibm Any ideas?