openbmc/openbmc #2918: Mirror rwfs to alt BMC chip

anoo1, 2018-02-15:
If the watchdog triggers the aspeed hw to boot from the alternate BMC chip due to a boot failure, the read-write data that was stored on the primary chip would not be available. This includes network configuration, so that BMC may not be reachable anymore, which defeats the purpose of switching chips to 'recover' from a failure.

Need to explore mirroring the read-write data stored in the rwfs volume to the alt chip, and also the time intervals at which this should be done.

anoo1, 2018-03-28:
Per Brad, look into a filesystem option/type that would allow mirroring, so when writing to one filesystem then kernel would write to a second one as well.

amboar, 2018-04-09:
It sounds like you want RAID 1, but do you *really* want RAID 1? RAID management of flash on a BMC seems like overkill.

rfrandse, 2018-04-24:
<https://gerrit.openbmc-project.xyz/10295> sync_manager: Add callback to do rsync
Closes  #2918 Mirror rwfs to alt BMC chip

closed 2018-05-08 by geissonator