openbmc/openbmc #2919: host-ipmi needs to remove legacy use of org.openbmc.control.Bmc

geissonator, 2018-02-15:
Here's the 2 instances I see.  New bmc reboot requests should go through BMC State (busctl set-property xyz.openbmc_project.State.BMC /xyz/openbmc_project/state/bmc0 xyz.openbmc_project.State.BMC RequestedBMCTransition s xyz.openbmc_project.State.BMC.Transition.Reboot"

```
/openbmc/phosphor-host-ipmid/
control_object_name = "/org/openbmc/control/bmc0";

/openbmc/openpower-host-ipmi-oem/
objname = "/org/openbmc/control/bmc0";
```

rfrandse, 2018-03-02:
<https://gerrit.openbmc-project.xyz/9317> Remove legacy use of org.openbmc.control.Bmc
Resolves  #2919 host-ipmi needs to remove legacy use of org.openbmc.control.Bmc

ngorugan, 2018-03-09:
@geissonator @tomjoseph83 Do we really need to remove legacy use of org.openbmc.control.Bmc form the component  "openbmc/openpower-host-ipmi-oem" ?
Is this component  being used in witherspoon image ?

geissonator, 2018-03-09:
If it's used in any system, we'll need to remove it because we're removing org.openbmc.control.Bmc  from all images.  If it's not used anywhere, then no, we wouldn't need to remove it.

ngorugan, 2018-03-09:
How to check if it is being used by some other systems or not ?

geissonator, 2018-03-09:
Ask @tomjoseph83 ?

closed 2018-03-12 by geissonator