openbmc/openbmc #2920: Update all OpenBMC runtime python scripts to support python 3

geissonator, 2018-02-15:
The new 2.4 yocto update coming in #2910 requires all python applications be compatible with python 3.  This issue tracks getting them all up to date.  The main area is the python scripts located in https://github.com/openbmc/skeleton.

A portion of these have been deprecated so be sure to check what's actually still being used on the systems before updating.

I know Balaji did a few of these, I believe we had to revert one of them though.

https://gerrit.openbmc-project.xyz/#/q/owner:%22Balaji+B+Rao+%253Cbalajibapu%2540gmail.com%253E%22

# revert for pyphosphor
https://gerrit.openbmc-project.xyz/#/c/8757/

amboar, 2018-02-17:
I haven't read up on the impacts of yocto 2.4, but this feels a little strange. Can we no longer install python 2.7 in the target image? I can understand the yocto build infrastructure requiring that python scripts hooking into it be python 3, but it feels surprising that it would force the requirement of python 3 in the target image

rfrandse, 2018-03-02:
<https://gerrit.openbmc-project.xyz/9389> Port to python 3
Partially resolves  #2920 Update all OpenBMC runtime python scripts to support python 3

geissonator, 2018-03-05:
I think @bradbishop ended up finding that we don't in fact have to update to python 3 within our image.  But at some point we'll want to, in order to take advantage of some of the new features in python 3.  But not required for yocto 2.4.

closed 2018-03-12 by geissonator

rfrandse, 2018-03-12:
<https://gerrit.openbmc-project.xyz/9446> Port to python 3
Partially resolves  #2920 Update all OpenBMC runtime python scripts to support python 3

rfrandse, 2018-03-12:
<https://gerrit.openbmc-project.xyz/9414> Port to python 3
Partially resolves  #2920 Update all OpenBMC runtime python scripts to support python 3

rfrandse, 2018-03-12:
<https://gerrit.openbmc-project.xyz/9455> Port to python 3
Resolves  #2920 Update all OpenBMC runtime python scripts to support python 3