openbmc/openbmc #2922: npm ERR! phantomjs-prebuilt@2.1.16 install: `node install.js` when building

gtmills, 2018-02-16:
Keep hitting this on one system

```
|   g++  -shared -pthread -rdynamic -m64  -Wl,-soname=binding.node -o Release/obj.target/binding.node -Wl,--start-group Release/obj.target/binding/src/binding.o Release/obj.target/binding/src/create_string.o Release/obj.target/binding/src/custom_function_bridge.o Release/obj.target/binding/src/custom_importer_bridge.o Release/obj.target/binding/src/sass_context_wrapper.o Release/obj.target/binding/src/sass_types/boolean.o Release/obj.target/binding/src/sass_types/color.o Release/obj.target/binding/src/sass_types/error.o Release/obj.target/binding/src/sass_types/factory.o Release/obj.target/binding/src/sass_types/list.o Release/obj.target/binding/src/sass_types/map.o Release/obj.target/binding/src/sass_types/null.o Release/obj.target/binding/src/sass_types/number.o Release/obj.target/binding/src/sass_types/string.o Release/obj.target/src/sass.a -Wl,--end-group
|   rm -rf "Release/binding.node" && cp -af "Release/obj.target/binding.node" "Release/binding.node"
| make: Leaving directory '/tmp/openbmc/work/all-openbmc-linux/phosphor-webui/1.0-r0/git/node_modules/gulp-sass/node_modules/node-sass/build'
| gyp info ok
| Installed to /tmp/openbmc/work/all-openbmc-linux/phosphor-webui/1.0-r0/git/node_modules/gulp-sass/node_modules/node-sass/vendor/linux-ppc64-46/binding.node
| npm info build /tmp/openbmc/work/all-openbmc-linux/phosphor-webui/1.0-r0/git/node_modules/gulp-sass
| npm info linkStuff gulp-sass@2.3.2
| npm info install gulp-sass@2.3.2
| npm info postinstall gulp-sass@2.3.2
| npm ERR! Linux 3.19.0-28.31-openpower1-generic
| npm ERR! argv "/tmp/openbmc/sysroots/ppc64le-linux/usr/bin/node" "/tmp/openbmc/sysroots/ppc64le-linux/usr/bin/npm" "--loglevel" "info" "--proxy=" "--https-proxy=" "install"
| npm ERR! node v4.6.1
| npm ERR! npm  v2.15.9
| npm ERR! code ELIFECYCLE
| 
| npm ERR! phantomjs-prebuilt@2.1.16 install: `node install.js`
| npm ERR! Exit status 1
| npm ERR!
| npm ERR! Failed at the phantomjs-prebuilt@2.1.16 install script 'node install.js'.
| npm ERR! This is most likely a problem with the phantomjs-prebuilt package,
| npm ERR! not with npm itself.
| npm ERR! Tell the author that this fails on your system:
| npm ERR!     node install.js
| npm ERR! You can get information on how to open an issue for this project with:
| npm ERR!     npm bugs phantomjs-prebuilt
| npm ERR! Or if that isn't available, you can get their info via:
| npm ERR!
| npm ERR!     npm owner ls phantomjs-prebuilt
| npm ERR! There is likely additional logging output above.
| npm info preuninstall phantomjs-prebuilt@2.1.16
| npm info uninstall phantomjs-prebuilt@2.1.16
| npm info postuninstall phantomjs-prebuilt@2.1.16
| 
| npm ERR! Please include the following file with any support request:
| npm ERR!     /tmp/openbmc/work/all-openbmc-linux/phosphor-webui/1.0-r0/git/npm-debug.log
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /tmp/openbmc/work/all-openbmc-linux/phosphor-webui/1.0-r0/temp/log.do_compile.26192
```

gtmills, 2018-02-16:
A lot online about this.. https://github.com/Medium/phantomjs/issues/730
`npm install phantomjs-prebuilt@2.1.16 --ignore-scripts` does fix the problem
https://gerrit.openbmc-project.xyz/#/c/8987/2

gtmills, 2018-02-19:
I have tried adding `bzip2-native` to the `DEPENDS_prepend` in phosphor-webui.bb but it did not work.

rfrandse, 2018-03-13:
<https://gerrit.openbmc-project.xyz/9530> Remove phantomjs
Resolves  #2922 npm ERR! phantomjs-prebuilt@2.1.16 install: `node install.js` when building

closed 2018-03-14 by geissonator