openbmc/openbmc #2925: GUI does not handle "Quiesced" state correctly.

gtmills, 2018-02-19:
The GUI displays the "Quiesced" state as "booting".

https://github.com/openbmc/phosphor-webui/blob/master/app/common/services/constants.js#L39

 
From https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/README.md 
"Quiesced means the host OS is in a quiesce state and the system should be checked for errors. "

The GUI should display that the host is "Quiesced"

iftekharuli, 2018-02-21:
After talking to our sponsor users and 4 different interviews with IBMers- we decided to change the text to "Standby"

gtmills, 2018-02-21:
I think of the "quiesced" state as an error state.. @geissonator care to comment here.

geissonator, 2018-02-21:
Yes, quiesced implied your host is in an error state.  The normal cause is a fail to boot the host for a variety of different reasons. An error event should be present indicating the reason.

gtmills, 2018-02-21:
I think "Failed", "Error", or "Quiesced" would be better.

iftekharuli, 2018-02-21:
I will ask the researchers in design team who worked with sponsor users and let you know.

zombiGoats, 2018-02-21:
The quiesced state should mean the system has just been paused/halted for some reason (that's what the term means), and the system **should** be checked for errors. This doesn’t definitively mean an error occurred. 

None of our tested users knew of the term or used the term quiesced. The term “stand by” was understood by users and they acknowledged if the system was in a stand by state for unknown reasons, they would look into potential issues.

gtmills, 2018-02-21:
It may not have been the best choice of wording on our part then. A "quiesced" state for us means the host is in an error state. IMO, "standby" is not a strong enough word for this error state.

iftekharuli, 2018-03-13:
After the discussion with design team, we decided to go with the word "Error" I will update the code

rfrandse, 2018-03-22:
<https://gerrit.openbmc-project.xyz/9650> Changing the text Quiesced to Error
Fixes  #2925 GUI does not handle "Quiesced" state correctly.

closed 2018-03-27 by geissonator