openbmc/openbmc #2926: kernel boots 10s faster on ubi0 vs. ubi4

geissonator, 2018-02-20:
Not sure if this is just the nature of the aspeed hardware but it's something I found when working on optimizing our boot times.

The kernel boot takes an extra 10 seconds (twice as long) to boot when the image it's booting from is on ubi4 vs. ubi0.

I believe our design is that ubi0 is one BMC flash chip and ubi4 is the other flash chip.

ubi0 boot:
```
[witherspoon]# dmesg
[    0.960000] aspeed-smc 1e620000.flash-controller: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
[    0.960000] aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000 - 0x2a000000 ] 128MB
[    0.960000] 3 ofpart partitions found on MTD device bmc
[    0.990000] Creating 3 MTD partitions on "bmc":
[    0.990000] 0x000000000000-0x000000060000 : "u-boot"
[    1.000000] 0x000000060000-0x000000080000 : "u-boot-env"
[    1.010000] 0x000000080000-0x000002000000 : "obmc-ubi"
[    1.020000] aspeed-smc 1e620000.flash-controller: mx25l25635e (32768 Kbytes)
[    1.030000] aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000 - 0x24000000 ] 32MB
[    1.030000] aspeed-smc 1e620000.flash-controller: CE2 window [ 0x24000000 - 0x2c000000 ] 128MB
[    1.050000] 3 ofpart partitions found on MTD device alt-bmc
[    1.050000] Creating 3 MTD partitions on "alt-bmc":
[    1.060000] 0x000000000000-0x000000060000 : "alt-u-boot"
[    1.060000] 0x000000060000-0x000000080000 : "alt-u-boot-env"
[    1.080000] 0x000000080000-0x000002000000 : "alt-obmc-ubi"
[    1.090000] aspeed-smc 1e630000.flash-controller: mx66l1g45g (131072 Kbytes)
[    1.100000] aspeed-smc 1e630000.flash-controller: CE0 window resized to 120MB (AST2500 HW quirk)
[    1.100000] aspeed-smc 1e630000.flash-controller: CE0 window [ 0x30000000 - 0x37800000 ] 120MB
[    1.120000] aspeed-smc 1e630000.flash-controller: CE0 window too small for chip 128MB
[    1.120000] aspeed-smc 1e630000.flash-controller: CE1 window [ 0x37800000 - 0x38000000 ] 8MB
[    1.890000] libphy: Fixed MDIO Bus: probed
[    1.890000] ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:23:e1 from chip
[    1.890000] ftgmac100 1e660000.ethernet: Using NCSI interface
[    1.890000] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
[    1.890000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.890000] ehci-platform: EHCI generic platform driver
[    2.100000] i2c /dev entries driver
[    2.100000] aspeed-i2c-bus 1e78a0c0.i2c-bus: i2c bus 2 registered, irq 26
[    2.100000] aspeed-i2c-bus 1e78a100.i2c-bus: i2c bus 3 registered, irq 27
[    2.100000] aspeed-i2c-bus 1e78a140.i2c-bus: i2c bus 4 registered, irq 28
[    2.100000] aspeed-i2c-bus 1e78a180.i2c-bus: i2c bus 5 registered, irq 29
[    2.100000] aspeed-i2c-bus 1e78a380.i2c-bus: i2c bus 9 registered, irq 30
[    2.100000] aspeed-i2c-bus 1e78a3c0.i2c-bus: i2c bus 10 registered, irq 31
[    2.100000] rtc-rv8803 11-0032: An alarm maybe have been missed.
[    2.200000] rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
[    2.210000] at24 11-0051: 8192 byte 24c64 EEPROM, writable, 1 bytes/write
[    2.210000] aspeed-i2c-bus 1e78a400.i2c-bus: i2c bus 11 registered, irq 32
[    2.210000] aspeed-i2c-bus 1e78a440.i2c-bus: i2c bus 12 registered, irq 33
[    2.210000] aspeed-i2c-bus 1e78a480.i2c-bus: i2c bus 13 registered, irq 34
[    2.210000] Driver for 1-wire Dallas network protocol.
[    2.210000] lm75 9-004a: hwmon0: sensor 'tmp275'
[    3.510000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[    3.990000] leds-pca955x 3-0060: leds-pca955x: Using pca9552 16-bit LED driver at slave address 0x60
[    4.000000] leds-pca955x 3-0060: gpios 238...253
[    4.010000] leds-pca955x 11-0060: leds-pca955x: Using pca9552 16-bit LED driver at slave address 0x60
[    4.020000] leds-pca955x 11-0060: gpios 222...237
[    4.020000] usbcore: registered new interface driver usbhid
[    4.030000] usbhid: USB HID core driver
[    4.080000] NET: Registered protocol family 10
[    4.080000] Segment Routing with IPv6
[    4.090000] NET: Registered protocol family 17
[    4.090000] 8021q: 802.1Q VLAN Support v1.8
[    4.100000] input: gpio-keys-polled as /devices/platform/gpio-keys-polled/input/input1
[    4.130000] ubi0: default fastmap pool size: 25
[    4.140000] ubi0: default fastmap WL pool size: 12
[    4.140000] ubi0: attaching mtd3
[    4.150000] ubi0: scanning is finished
[    4.170000] ubi0: attached mtd3 (name "obmc-ubi", size 31 MiB)
[    4.170000] ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    4.190000] ubi0: min./max. I/O unit sizes: 1/256, sub-page size 1
[    4.190000] ubi0: VID header offset: 64 (aligned 64), data offset: 128
[... 122 lines cut ...]
[    4.370000] ubi4: available PEBs: 46, total reserved PEBs: 458, PEBs reserved for bad PEB handling: 0
[    4.370000] ubi4: background thread "ubi_bgt4d" started, PID 534
[   15.170000] block ubiblock4_0: created from ubi4:0(rofs-03f1b7ca)
[   15.170000] console [netcon0] enabled
[   15.170000] netconsole: network logging started
[   15.170000] input: gpio-keys as /devices/platform/gpio-keys/input/input2
[   15.170000] rtc-rv8803 11-0032: setting system clock to 2018-02-20 16:36:57 UTC (1519144617)
[   15.170000] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
[   15.170000] devtmpfs: mounted
[   15.170000] Freeing unused kernel memory: 280K
[   15.170000] This architecture does not have kernel memory protection.
[   17.340000] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 543
[   17.340000] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rwfs"
[   17.340000] UBIFS (ubi0:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   17.340000] UBIFS (ubi0:2): FS size: 3597440 bytes (3 MiB, 55 LEBs), journal size 719488 bytes (0 MiB, 11 LEBs)
[   17.340000] UBIFS (ubi0:2): reserved for root: 0 bytes (0 KiB)
[   17.340000] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID E59D6927-10A2-47FF-8682-D54D5F58127E, small LPT model
[   19.530000] systemd[1]: Failed to insert module 'autofs4': No such file or directory
[   19.530000] systemd[1]: systemd 232 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN)
```
```
Feb 20 16:38:56 witherspoon systemd[1]: Startup finished in 19.238s (kernel) + 1min 55.160s (userspace) = 2min 14.399s.
```

FYI @anoo1 @mdmillerii

amboar, 2018-02-20:
There is an issue with the flash controller (or driver, I don't quite recall) where the performance of the second chip does not match the first. @legoater can provide more info here.

legoater, 2018-02-21:
yes. the flash controller is most certainly not tuned for fast reads on the second chip. I don't think this is covered by the driver. To be added on the Linux TODO list.

legoater, 2018-02-21:
new Linux issue on the topic : https://github.com/openbmc/linux/issues/143

closed 2019-04-27 by stale[bot]