openbmc/openbmc #2927: Hitting the "refresh" button on the firmware page does not refresh the firmware information.

gtmills, 2018-02-20:
While on the firmware tab, <bmc>/#/configuration/firmware clicking "refresh" does not refresh the firmware information. After clicking "refresh" I can see in the journal the only Rest calls are for state, logging, and network. There should be a rest call for /xyz/openbmc_project/software/enumerate
```
Feb 20 17:31:37 witherspoon phosphor-gevent[1842]: ::da:c376:8e91:f402%1101898292 - - [2018-02-20 17:31:37] "GET /xyz/openbmc_project/state/host0 HTTP/1.1" 200 629 0.080919
Feb 20 17:31:37 witherspoon phosphor-gevent[1842]: ::da:c376:8e91:f402%1101898292 - - [2018-02-20 17:31:37] "GET /xyz/openbmc_project/logging/enumerate HTTP/1.1" 200 319 0.156479
Feb 20 17:31:37 witherspoon phosphor-gevent[1842]: ::da:c376:8e91:f402%1101898292 - - [2018-02-20 17:31:37] "GET /xyz/openbmc_project/network/enumerate HTTP/1.1" 200 2037 0.316969
```

camvanng, 2018-03-22:
Other pages also do not refresh.

rfrandse, 2018-03-22:
<https://gerrit.openbmc-project.xyz/9658> Fix refresh on system & sensors overview pages
Partially resolves  #2927 Hitting the "refresh" button on the firmware page does not refresh the firmware information.

iftekharuli, 2018-03-22:
This is correct. Instead of page refresh we developed a better method of just to call the rest method needed to refresh the page data only.  This has been coded for log page and as per discussion, the dev team will code the rest of the pages. This code for log page exists here- https://gerrit.openbmc-project.xyz/#/c/9645/

One point I want to emphasis on coding this to other pages is to remember destroy you listener call  at the end of the call otherwise you will encounter chain call issues. All in example code above.

gtmills, 2018-03-29:
I have reopened the issue, this issue is not fixed. It closed automatically due to the `Partially resolves` in  openbmc/phosphor-webui@23217da. The tooling apparently does not recognize the difference between `Partially resolves` and `resolves`.

closed 2018-04-12 by geissonator