openbmc/openbmc #2928: No "Image State" is shown for an image with "Activation" "Activating"

gtmills, 2018-02-20:
[image]
```
    "/xyz/openbmc_project/software/511c40be": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Activating", 
      "Path": "/tmp/images/511c40be", 
      "Progress": 30, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active", 
      "Version": "ibm-v2.0-0-r35-0-g7c7c1af", 
      "associations": [
        [
          "inventory", 
          "activation", 
          "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
      ]
    },
```

gtmills, 2018-02-20:
Should also check if other activation values like "Failed" and "NotReady" appear.
https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Software/Activation.interface.yaml#L17

rfrandse, 2018-02-23:
<https://gerrit.openbmc-project.xyz/9165> Add additional "Image States"
Resolves  #2928 No "Image State" is shown for an image with "Activation" "Activating"

closed 2018-03-01 by geissonator