openbmc/openbmc #2929: AST2500 Evaluation Ethernet With Realtek RTL8211F

ThaiThu, 2018-02-21:
Hi all.

Our BMC is using RTL8211F as PHY device.
We already enable MAC0 in device tree file with ncsi as belo:

&mac0 {
        status = "okay";

        use-ncsi;

        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_rgmii1_default &pinctrl_mdio1_default>;
};

When we "ifconfig eth0 up", we receive the error as below:

[ 4151.150000] ftgmac100 1e660000.ethernet eth0: NCSI: No channel found with link
[ 4151.160000] ftgmac100 1e660000.ethernet eth0: NCSI interface down

Is there any hint or recommendation that I can try, please !! 
Thank you very much !!

amboar, 2018-02-21:
Hi @ThaiThu, can you link us to the kernel tree you are using? We will need a little bit more context to debug your issue.

ThaiThu, 2018-02-22:
Hi @amboar , the kernel we are using is:
https://github.com/openbmc/linux

And the device tree is /arch/arm/boot/dts/aspeed-ast2500-evb.dts

Thank you @amboar

ThaiThu, 2018-02-22:
Hi @amboar , we see that the REALTEK PHY Driver is already configured in the defconfig file

ThaiThu, 2018-02-22:
Update:

Sorry, We found out that the Realtek Phy Device's reset pin is in reset status and is controlled by our BMC.
But we now have another problem is, we are able to ping BMC from notebook, but imposible vice versus.
BMC is not able to ping out.

Thank you !!

amboar, 2018-02-22:
@ThaiThu great to hear you found the issue with the reset pin. I'd like to help with your follow up question as well, but I think that's best done on the mailing list. We try to keep issues in the issue tracker focused on one problem, otherwise it becomes hard to know when to close an issue.

If you haven't already, sign yourself up here: https://lists.ozlabs.org/listinfo/openbmc

The list is fairly low volume and people are happy to help. Hope to see your post there!

closed 2018-02-22 by amboar