openbmc/openbmc #2931: Stop ethernet port fail over on witherspoon

geissonator, 2018-02-21:
The witherspoon system owner has requested a mechanism to prevent the BMC from failing over to the other ethernet port if it detects a network drop on the one it's using.

@sammj has a patch on the mailing list via http://patchwork.ozlabs.org/patch/873610/ for the kernel support.  He also has a user space application that can be run to configure this using his new driver in https://github.com/sammj/ncsi-netlink

This story is to integrate this all into the witherspoon layer and set the NCSI to not do fail overs to the other port.

sammj, 2018-02-23:
Proper V1 now sent to netdev: http://patchwork.ozlabs.org/patch/876963/

sammj, 2018-03-05:
V2 applied to net-next: http://patchwork.ozlabs.org/patch/881249/

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9620> Network: Bring libnl into the openbmc image
Partilally Resolves  #2931 Stop ethernet port fail over on System

sammj, 2018-03-29:
FWIW https://github.com/openbmc/openbmc/commit/7a5340bd5014a9af06b2de9443479100acd02eaa only introduces the libnl-genl dependency, the actual failover setting is controlled in https://gerrit.openbmc-project.xyz/#/c/9670/ etc

dkodihal, 2018-03-29:
https://gerrit.openbmc-project.xyz/#/c/9670/ only partially resolved this.

ratagupt, 2018-04-01:
This issue is addressing to not allow the failover,There was not a requirement to make it configurable.
However we will make it configurable as a follow on feature if someone requests it.

closed 2018-04-05 by geissonator