openbmc/openbmc #2932: Refactor Set LAN configuration IPMI command implementation

tomjoseph83, 2018-02-21:
The current implementation of IPMI stack on OpenBMC uses the Set Channel Access command to apply the network settings. This is not a standard way of applying the network settings, as other BMC implementations do not implement this. So this can cause a disconnect between user expectations and our implementation.

There was brainstorming done on this in the mailing list. Couple of suggestions came up:

a) Wait for 10 secs (or a specified time) after the Set Complete operation, before applying settings, to
allow multiple changes to be grouped into a single reconfiguration. This would involve changes only on the ipmi stack on OpenBMC.

b) Allow multiple parameter change on one invocation with ipmitool.
 ipmitool [...] lan set 1 ipsrc static ipaddr 192.168.0.2 netmask 255.255.255.0
This would change all the parameters under one transition of the set-complete flag.

Adding the links to the discussions on the mailing list.
https://lists.ozlabs.org/pipermail/openbmc/2017-December/010125.html
https://lists.ozlabs.org/pipermail/openbmc/2018-February/010769.html

vmauery, 2018-02-22:
Nobody at Intel is currently working on this, but we do support this path forward.

rfrandse, 2018-03-07:
<https://gerrit.openbmc-project.xyz/9434> SetLan: Apply the network changes once the timer expires
Resolves  #2932 Refactor Set LAN configuration IPMI command implementation

closed 2018-03-20 by geissonator