openbmc/openbmc #2938: Reduce retries in witherspoon-pseq-monitor when needed gpio not available

geissonator, 2018-02-22:
In our move to the 4.13 kernel, this popped up (over and over and over :)  )

Probably be good to try a few times, log an error, and die.

```
Feb 22 13:33:58 witherspoon witherspoon-pseq-monitor[1997]: Failed opening GPIO device
Feb 22 13:33:58 witherspoon witherspoon-pseq-monitor[1997]: The operation failed internally.
Feb 22 13:33:59 witherspoon witherspoon-pseq-monitor[1997]: Failed opening GPIO device
Feb 22 13:33:59 witherspoon witherspoon-pseq-monitor[1997]: The operation failed internally.
Feb 22 13:34:00 witherspoon witherspoon-pseq-monitor[1997]: Failed opening GPIO device
Feb 22 13:34:00 witherspoon witherspoon-pseq-monitor[1997]: The operation failed internally.
```

rfrandse, 2018-02-28:
<https://gerrit.openbmc-project.xyz/9253> Give up polling GPIOs after too many fails
Resolves  #2938 Reduce retries in witherspoon-pseq-monitor when needed gpio not available

spinler, 2018-03-21:
As this can't happen on a properly built witherspoon driver. going to just close this.

closed 2018-03-21 by spinler