openbmc/openbmc #2939: Have the REST upload return the id

gtmills, 2018-02-22:
The REST server should return the id (e.g. 36d083d1) of the uploaded image (e.g. obmc-phosphor-image-witherspoon.ubi.mtd.tar).
The REST server uploads the file to /tmp/images/, then the phosphor-version-software-manager (notified by inotify that a new file has been uploaded) untars the file, calculates the ID, and creates a new D-Bus object.
The REST server would need to listen for the creation of a new software D-Bus object and return the id. 
This is RACE condition prone, because if while uploading an image, a user downloads Via TFTP or copies to /tmp/images a different image, the incorrect id could be returned.

amboar, 2018-02-23:
For one we need to make sure that unique file names are being created in `/tmp/images`. Assuming that, we can publish (ID, path) or (ID, hash(file)) in the DBus object. That would give the webserver the ability to find the right ID for the request it is servicing.

gtmills, 2018-02-23:
The rest server can get the ID from either the path we publish (e.g. /tmp/images/36d083d1) or from the D-Bus object itself (e.g. /xyz/openbmc_project/software/36d083d1). The rest server has no idea which D-Bus object associates to the file (e.g. obmc-phosphor-image-witherspoon.ubi.mtd.tar) it uploaded to `/tmp/images/` . The file uploaded to `/tmp/images/` is untarred, the id is calculated from the version, then it is deleted. There is no guarantee that file name (e.g. obmc-phosphor-image-witherspoon.ubi.mtd.tar) is unique even if that name was published, this is due to 
"It is possible for the user to choose the uploaded file's remote name:

```curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" \
    -X PUT -T foo https://${bmc}/upload/image/bar.```  "

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9594> rest_dbus: do_upload: Add version id error code
Fixes  #2939 Have the REST upload return the id

closed 2018-04-11 by geissonator