openbmc/openbmc #2940: [Witherspoon] OP920 BMC reports 0 for target speed of fans

cdeadmin, 2018-02-23:


spinler, 2018-02-23:
There was recently a code change to add the Control.FanPwm interface to sensor objects whose hwmon driver exposed a 'pwm' sysfs file.  Unfortunately, the MAX31785 controller exposed both a 'pwm' file and a 'target' file, so both the FanPwm and FanSpeed interfaces were created, each providing a Target property.  On REST calls there is no way to specify the interface, so the property for the Pwm interface was returned, which was unused as RPM mode was being used.

rfrandse, 2018-02-23:
<https://gerrit.openbmc-project.xyz/9173> Set Witherspoon hwmon target mode
Resolves  #2940 BMC reports 0 for target speed of fans on systems with MAX31785 fan controller

spinler, 2018-02-23:
To fix this, I added some smarts to be able to select which interface to create, and to have a decent default that will only try one or the other.

closed 2018-02-27 by geissonator