openbmc/openbmc #2945: Determine "Functional" images by association not by priority 0

gtmills, 2018-02-23:
The functional, or running, images should be determined by looking at the functional association, not by looking at priority 0.
An image could be functional but not priority 0, this happens after activating a new image. This change is important because the functional image cannot be deleted and we hide the delete button for the functional image on the GUI so if it is incorrect we are hiding the delete button on the wrong image.

https://github.com/openbmc/phosphor-webui/blob/master/app/common/services/api-utils.js#L649

gtmills, 2018-02-23:
Also, have cases where there is no priority 0.

rfrandse, 2018-03-02:
<https://gerrit.openbmc-project.xyz/9179> Determine "Functional" images by association
Resolves  #2945 Determine "Functional" images by association not by priority 0