openbmc/openbmc #2946: [CRITICAL] OpenBMC corrupts itself after random number of standby power cycles

madscientist159, 2018-02-23:
After a random number of standby power cycles, OpenBMC no longer boots.  This has been observed on at least one shipped system and also internally in our lab.  The issue appears to be the rw partition running out of space, thus preventing rootfs mount.

Full log with directory listing of full rw partition:

```
 
DRAM Init-V10-DDR4
0abc1-4Gb-Done
Read margin-DL:0.3647/DH:0.3803 CK (min:0.30)

U-Boot 2016.07 (Jan 22 2018 - 03:12:32 -0600)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  2  1  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    2082360 Bytes = 2 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   ba526566291daeab4204aec6851f1cc81c33f93a
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x20283e9c
     Data Size:    1585141 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   f106b6d05fa0332ac0dfbbe22ad993ae1e0a2c54
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x2027c854
     Data Size:    30085 Bytes = 29.4 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   524d09f30aaa28d79b53f99edf3c8faefef9cf22
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x2027c854
   Loading Kernel Image ... OK
   Loading Ramdisk to 9ea15000, end 9eb97ff5 ... OK
   Loading Device Tree to 9ea0a000, end 9ea14584 ... OK

Starting kernel ...
[... 715 lines cut ...]
-rw-r--r--    1 root     root           230 Feb 23  2018 [0;0m89061[0m
-rw-r--r--    1 root     root           230 Feb 23  2018 [0;0m89062[0m
-rw-r--r--    1 root     root           228 Feb 23  2018 [0;0m89063[0m
-rw-r--r--    1 root     root           228 Feb 23  2018 [0;0m89064[0m
-rw-r--r--    1 root     root           228 Feb 23  2018 [0;0m89065[0m

/run/initramfs/rw/cow/var/lib/phosphor-state-manager:
total 1
-rw-r--r--    1 root     root           305 Feb 21  2018 [0;0mrequestedHostTransition[0m

/run/initramfs/rw/cow/var/lib/systemd:
total 1
drwxr-xr-x    2 root     root             0 Jan 19  2018 [1;34mcatalog[0m
-rw-r--r--    1 997      994              0 Feb 23  2018 [0;0mclock[0m
drwxr-xr-x    2 root     root             0 Feb 23  2018 [1;34mcoredump[0m
-rw-------    1 root     root           512 Jan 19  2018 [0;0mrandom-seed[0m

/run/initramfs/rw/cow/var/lib/systemd/catalog:
total 13
-rw-r--r--    1 root     root         12601 Jan 19  2018 [0;0mdatabase[0m

/run/initramfs/rw/cow/var/lib/systemd/coredump:
total 0
/ # 
```

madscientist159, 2018-02-23:
Possible mitigation: reformat the rw partition on failure to create `run/initramfs/rw/work` ?

geissonator, 2018-02-23:
One thing we realized before shipping was that enabling core dumps caused more issues then it was worth.  We've been working through that in master but for our shipment, we pulled in https://gerrit.openbmc-project.xyz/#/c/7949/.  I assume those big dumps taking up all the space are core one's?  May be best to clear the filesystem and pull in https://gerrit.openbmc-project.xyz/#/c/7949/.

- Need to get to root cause on what's causing the core dumps.
- Need to do the math and figure out how much space you have available for dumps (with full error logs included).

@ojayanth and @anoo1 have been working on that for us.  We recently bumped up the size allocated to the write partitions.

Just doing a rm -rf on /var/lib/phosphor-debug-collector/* and /var/lib/phosphor-logging/* should help (after you copy off some for debug).

madscientist159, 2018-02-23:
@geissonator Good to know.  What was most surprising was that filling up rw prevented boot; core dumps are only one way to fill up the rw partition.  I am looking for a more general solution that will allow recovery if the rw partition renders the BMC unbootable for any reason. 

Looking through the file tree it looks like `/etc` needs to be preseved but the rest of the partition can be wiped as a last resort?

geissonator, 2018-02-23:
Yeah, some of the user settings are stored in /var/ I believe so those would get lost on a reset if the user has changed anything from the defaults, but usually the most important thing is the network config file, which is in etc.  Systemd has some configurations parameters for core dumps, where you can tell it max sizes and how much space it needs to leave available in the filesystem it's writing too, but you're right, lots of things can fill up the rw partition.  I still have this commit out in gerrit, https://gerrit.openbmc-project.xyz/#/c/7844/, to disable the wtmp logging.

madscientist159, 2018-02-24:
Recovery could be as simple as running this if the root mount fails:

```
cd /run/initramfs/rw/cow
ls | grep -v etc | grep -v var | xargs rm -rf
rm -rf var/lib/phosphor*
reboot -f
```

We'll probably add this as a failsafe in our builds.

mdmillerii, 2018-03-10:
The standard obmc-phosphor-initfs initrd has a factory reset that copies the whitelisted content to ram, erases the rwfs, then copies the files back.  Currently this is triggered from an option passed either via kernel command line or one of 2 u-boot env variables (one of which is cleared by another systemd service unit each boot).

closed 2019-04-27 by stale[bot]