openbmc/openbmc #2947: IPMI: Populate manufacturer ID and name

shenki, 2018-02-26:
We should populate the manufacturer ID and name:

Current behaviour:
```
$ ipmitool [...] -H ozrom2-bmc mc info
[...]
Manufacturer ID           : 0
Manufacturer Name         : Unknown
```
Expected behaviour:
```
$ ipmitool [...] -H ozrom2-bmc mc info
[...]
Manufacturer ID           : <number>
Manufacturer Name         : OpenBMC
```

klema1, 2018-03-16:
Does anyone know who assigns the number for the Manufacturing ID?  We are attempting to use "mc info" operation to identify which BMC Platform for https://github.com/open-power/skiboot/issues/159

stewart-ibm, 2018-03-16:
In talking to @tomjoseph83 on internal slack, Romulus will need something like https://gerrit.openbmc-project.xyz/#/c/8041 which sets the manufacturer ID for 'mc info'.

e.g. witherspoon currently provides this info:
```
Device ID                 : 0
Device Revision           : 0
Firmware Revision         : 2.01
IPMI Version              : 2.0
Manufacturer ID           : 42817
Manufacturer Name         : Unknown (0xA741)
Product ID                : 16975 (0x424f)
Product Name              : Unknown (0x424F)
Device Available          : yes
Provides Device SDRs      : yes
Additional Device Support :
    Sensor Device
    SEL Device
    FRU Inventory Device
    Chassis Device
Aux Firmware Rev Info     : 
    0x00
    0x00
    0x00
    0x00
```

It may be an idea for this to be a default for OpenBMC rather than relying on each platform to set something sensible?

AlexanderAmelkin, 2018-04-06:
@klema1 the manufacturer IDs in IPMI are IANA Private Enterprise Numbers, same as in SNMP.
Assigned free of charge by IANA.org here:
http://pen.iana.org/pen/PenApplication.page

The list of current assignments is available online here:
https://www.iana.org/assignments/enterprise-numbers/enterprise-numbers
(the top of trunk `ipmitool` now fetches, parses and incorporates this list at build time)

AlexanderAmelkin, 2018-04-06:
@stewart-ibm with all due respect to IBM and your contribution to development of OpenBMC, I nonetheless think that defaulting to "IBM Platform Firmware Division" as manufacturer ID is wrong.
I'd propose registering a separate PEN specifically for OpenBMC. I guess this PEN should be registered to LF Projects, LLC as the legal entity behind the OpenBMC community.

AlexanderAmelkin, 2018-04-06:
@stewart-ibm I've just discovered that there is PEN 49871 registered to @williamspatrick for "OpenBMC Project". I personally think that in the long run it is very wrong that the PEN is registered to a person and his personal e-mail domain instead of a non-profit organization. However, we could still switch to that PEN as default.

stewart-ibm, 2018-04-13:
@AlexanderAmelkin I'm fine with an OpenBMC one, that's kind of what I'd expect everywhere TBH.

Anything except zero or a random number is good :)

AlexanderAmelkin, 2018-04-13:
@stewart-ibm, the ownership of the OpenBMC PEN is what bothers me. I hope @williamspatrick will stay with the community for a long time, but I've seen projects where holders of crucible pieces abandoned them and it has always took a lot of effort for the rest of the teams to regain control. Hence, my proposal is to think about it in advance and agree with Patrick on changing management contact for OpenBMC PEN to something neutral. Then change openbmc to that PEN.

closed 2019-04-27 by stale[bot]

stewart-ibm, 2019-04-29:
as far as I'm aware, this has not actually been fixed yet?