openbmc/openbmc #2948: [kernel-4.13] boot hangs for 100+ seconds in ISTEPs

mine260309, 2018-02-27:
This is to track an issue found on dev-4.13 kernel, which is reproduced on both P8 Palmetto (@shenki) and P9 Romulus.

With dev-4.13 kernel, booting host is OK, but it takes much longer to boot.
From log, it hangs for more than 100 seconds in ISTEP21.1 and after ISTEP21.3

```
...
 31.24767|ISTEP 20. 1 - host_load_payload
 31.61207|ISTEP 20. 2 - host_load_hdat
 32.82990|ISTEP 21. 1 - host_runtime_setup
176.45759|htmgt|OCCs are now running in ACTIVE state
179.41235|ISTEP 21. 2 - host_verify_hdat
179.39749|ISTEP 21. 3 - host_start_payload
[  298.313074164,5] OPAL skiboot-v5.10-rc1 starting...
[  298.313077708,7] initial console log level: memory 7, driver 5
...
```

mine260309, 2018-02-27:
From hostboot perspective, added some logs in hostboot ISTEP 21.1, and it shows that it hangs in `IPMI::sendrecv(IPMI::get_power_limit(), cc, len, data);` in `ipmidcmi.C`, that it takes long time to send and receive the ipmi command.

From BMC, enabling IPMI traces and it shows that `ipmid` does not receive the ipmi command during the hang time; instead it receives the command after the hang.

I tried to add `CONSOLE::displayf` log in `BTMessage::phy_xmit(void)` and `BTMessage::recv(void)`, however, with the logs, the issue does not occur anymore.

So it seems to indicate that there is something wrong with LPC, because `CONSOLE::displayf` write logs to VUART via LPC as well.

geissonator, 2018-02-27:
Yeah, I saw this same issue on a witherspoon testing this out:

```
143.45765|ISTEP 20. 2 - host_load_hdat
146.48752|ISTEP 21. 1 - host_runtime_setup
225.75044|htmgt|OCCs are now running in ACTIVE state
225.75141|ISTEP 21. 2 - host_verify_hdat
225.79079|ISTEP 21. 3 - host_start_payload
[  241.378055966,5] OPAL skiboot-v5.9.5 starting...

```

mine260309, 2018-02-28:
Enabled ipmid trace and compare the ipmid logs.
[ipmid_4.10.log](https://github.com/openbmc/openbmc/files/1765836/ipmid_4.10.log)
[ipmid_4.13.log](https://github.com/openbmc/openbmc/files/1765837/ipmid_4.13.log)

Observasions:
1. In 4.13, it takes much longer time to receive the ipmi commands;
2. In 4.13, systemd-journald seems to cache the log, delay it and prints the log all together;
 If I stop `phosphor-ipmi-host.service` and manually run `ipmid`, the logs prints to console, I can see the logs are printed without the "cache and delay", but it is much slower than 4.10, something like one ipmi command per second.

tomjoseph83, 2018-03-01:
The  printf/fprintf statements in host-ipmid(or any other component in OpenBMC) cannot be relied upon based on the timestamps it  displays. The timestamps and the ordering of the printf statements are haphazard. 

It needs to be moved to sd_journal_print if the timing sequence needs to be correctly understood.

mine260309, 2018-03-02:
Regarding console log:
>I tried to add CONSOLE::displayf log in BTMessage::phy_xmit(void) and BTMessage::recv(void), however, with the logs, the issue does not occur anymore.

I make below changes in hostboot, that try to log the timestamps of each `IpmiDD::send()`: [patch-timestamp-dd-and-after-dcmi.patch.txt](https://github.com/openbmc/openbmc/files/1773794/patch-timestamp-dd-and-after-dcmi.patch.txt)

1. If I disable the `CONSOLE::display()` in `ipmidd.C`, the issue occurs.
The log: [hostboot_romulus-4.13.timestamp-only-after-dcmi.log](https://github.com/openbmc/openbmc/files/1773798/hostboot_romulus-4.13.timestamp-only-after-dcmi.log)
2. If I enable the `CONSOLE::display()` in `ipmidd.C`, the issue is gone.
The log: 
[hostboot_romulus-4.13.timestamp-ipmidd-and-after-dcmi.log](https://github.com/openbmc/openbmc/files/1773799/hostboot_romulus-4.13.timestamp-ipmidd-and-after-dcmi.log)

From the logs, we can see that if the issue occurs, the `IpmiDD::send()` is called rather slowly, most of the IPMI commands are sent once per second; but each command is sent quickly as normal case.

This seems to indicate that the IPMI commands are queued in hostboot, delayed and sent very slowly.

geissonator, 2018-03-02:
We had some discussions with the host team but didn't find much. They don't know much about the BMC hw they're talking to until they start talking to the BMC, so having them do any hw inits specific to the AST2500 is a no go for them.  @eddiejames volunteered to see if he could find anything on the BMC side that we need.

mine260309, 2018-03-06:
I have got some clue that the issue is related to lpc clock.

In 4.13 we have new clock driver for aspeed, and lpc uses the clock.

If I comment out the clock related code:
```
diff --git a/drivers/misc/aspeed-lpc-ctrl.c b/drivers/misc/aspeed-lpc-ctrl.c
index a024f80..6fafdf8 100644
--- a/drivers/misc/aspeed-lpc-ctrl.c
+++ b/drivers/misc/aspeed-lpc-ctrl.c
@@ -237,16 +237,16 @@ static int aspeed_lpc_ctrl_probe(struct platform_device *pdev)
                return -ENODEV;
        }

-       lpc_ctrl->clk = devm_clk_get(dev, NULL);
-       if (IS_ERR(lpc_ctrl->clk)) {
-               dev_err(dev, "couldn't get clock\n");
-               return PTR_ERR(lpc_ctrl->clk);
-       }
-       rc = clk_prepare_enable(lpc_ctrl->clk);
-       if (rc) {
-               dev_err(dev, "couldn't enable clock\n");
-               return rc;
-       }
+//     lpc_ctrl->clk = devm_clk_get(dev, NULL);
+//     if (IS_ERR(lpc_ctrl->clk)) {
+//             dev_err(dev, "couldn't get clock\n");
+//             return PTR_ERR(lpc_ctrl->clk);
+//     }
+//     rc = clk_prepare_enable(lpc_ctrl->clk);
+//     if (rc) {
+//             dev_err(dev, "couldn't enable clock\n");
+//             return rc;
+//     }

        lpc_ctrl->miscdev.minor = MISC_DYNAMIC_MINOR;
        lpc_ctrl->miscdev.name = DEVICE_NAME;
```
And enable LPC (and other clocks) manually: `devmem 0x1e6e200c 32 0xF9E47E8B` (the value is taken from 4.10 kernel), the issue is gone.

If I only manually set `0x1e6e200c` without commenting clock related code in `aspeed-lpc-ctrl.c`, the issue still remains.

So we probably have something wrong with LPC clock.

mine260309, 2018-03-06:
This patch also "fixes" the issue:
```
diff --git a/drivers/clk/clk-aspeed.c b/drivers/clk/clk-aspeed.c
index b67ba14..72ae382 100644
--- a/drivers/clk/clk-aspeed.c
+++ b/drivers/clk/clk-aspeed.c
@@ -224,25 +224,25 @@ static int aspeed_clk_enable(struct clk_hw *hw)

        spin_lock_irqsave(gate->lock, flags);

-       if (gate->reset_idx >= 0) {
-               /* Put IP in reset */
-               regmap_update_bits(gate->map, ASPEED_RESET_CTRL, rst, rst);
-
-               /* Delay 100us */
-               udelay(100);
-       }
+//     if (gate->reset_idx >= 0) {
+//             /* Put IP in reset */
+//             regmap_update_bits(gate->map, ASPEED_RESET_CTRL, rst, rst);
+//
+//             /* Delay 100us */
+//             udelay(100);
+//     }

        /* Enable clock */
        enval = (gate->flags & CLK_GATE_SET_TO_DISABLE) ? 0 : clk;
        regmap_update_bits(gate->map, ASPEED_CLK_STOP_CTRL, clk, enval);

-       if (gate->reset_idx >= 0) {
-               /* A delay of 10ms is specified by the ASPEED docs */
-               mdelay(10);
-
-               /* Take IP out of reset */
-               regmap_update_bits(gate->map, ASPEED_RESET_CTRL, rst, 0);
-       }
+//     if (gate->reset_idx >= 0) {
+//             /* A delay of 10ms is specified by the ASPEED docs */
+//             mdelay(10);
+//
+//             /* Take IP out of reset */
+//             regmap_update_bits(gate->map, ASPEED_RESET_CTRL, rst, 0);
+//     }

        spin_unlock_irqrestore(gate->lock, flags);
```

The aspeed data sheet says below about the reset:
> The sequence must do once whenever the engine is started from the clock stopped state.

But the LPC clock is started by default, so it does the reset -> enable clock -> disable reset sequence in started state, that may cause this issue.

So the code of `aspeed_clk_enable()` may be changed to check if the clock is enabled or not, if it's enabled, just do nothing.

eddiejames, 2018-03-06:
Great find Lei. I'll get a patch on the mailing list and see what people think of it.

mine260309, 2018-03-30:
The issue is fixed in dev-4.13 with https://github.com/openbmc/linux/commit/b9d07b242af0251e890a6919c098cb372d51571a

closed 2018-03-30 by mine260309