openbmc/openbmc #2949: cross-localedef-native: Build fail

rpking42, 2018-02-28:
I'm getting the following fail on an up-to-date Fedora 27 system when running `bitbake obmc-phosphor-image` with no `TEMPLATECONF`. I have not tried any other configurations, but this same recipe works for me in a Fedora 23 VM.
```
Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-27"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-poky         
meta-oe           
meta-networking   
meta-python       
meta-virtualization 
meta-phosphor     = "master:a36894f97204c466ba00606bfa44e9175c747b11"
```

Error Snippet:
```
In file included from glibc/locale/locale.h:145:0,
                 from ../git/localedef/include/locale.h:1,
                 from /usr/include/libintl.h:103,
                 from ../git/localedef/include/libintl.h:2,
                 from glibc/locale/programs/charmap.c:24:
glibc/locale/xlocale.h:27:16: error: redefinition of ‘struct __locale_struct’
 typedef struct __locale_struct
                ^~~~~~~~~~~~~~~
```

Full Log (renamed with a .txt extension to make GitHub happy):
[log.do_compile.26973.txt](https://github.com/openbmc/openbmc/files/1765664/log.do_compile.26973.txt)

geissonator, 2018-02-28:
Any thoughts on this one @bradbishop ?  I'm wondering if the new compiler in fedora 27 used for compiling the compiler is catching something?  I also wonder if your update to 2.3 will fix this?  https://gerrit.openbmc-project.xyz/#/c/8792/

mine260309, 2018-02-28:
@amboar had a temp patch for similar issue:
https://raw.githubusercontent.com/amboar/openbmc-bootstrap/master/ubuntu/17.10/0001-localedef-Hacks-to-work-on-Ubuntu-17.10.patch

Maybe try it to see if it works on Fedora 27 as well.

rpking42, 2018-02-28:
That patch for Ubuntu worked for me as well. I'm able to build and run in qemu. If a final fix is available I can try that in the future.

I didn't attempt to run with that gerrit change, because it looks like a pretty major change and it also looks like there is a conflict.

amboar, 2018-02-28:
Hmm, can people contribute their hacks back to openbmc-bootstrap if they have got openbmc building on a new distro? It will be handy for others.

I should add a bit more documentation.

rpking42, 2018-03-14:
I saw that the 2.3 update went into master, so I updated and I was able to build directly off of master on Fedora 27, so it appears this issue is resolved.

amboar, 2018-03-14:
Between the work-around patch and master I think people have enough information to move forward.

closed 2018-03-14 by amboar