openbmc/openbmc #2954: xyz.openbmc_project.Software.BMC.Updater.service core dump on DeleteAll.

gtmills, 2018-02-28:
Seeing this on the latest master. 
Steps to reproduce
1) Have 2 "active" BMC images. 
2) Call the BMC software DeleteAll (using the curl command calls both the Host and BMC software DeleteAll which is fine but if you want to call just the BMC software deleteAll you must use the busctl or :3000 ). 
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" \
     -X POST https://${bmc}/xyz/openbmc_project/software/action/deleteAll \
     -d "{\"data\": [] }"
```

```Feb 21 04:05:21 witherspoon kernel: UBIFS (ubi8:1): un-mount UBI device 8
Feb 21 04:05:21 witherspoon kernel: UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
Feb 21 04:05:21 witherspoon kernel[1038]: [127298.880000] UBIFS (ubi8:1): un-mount UBI device 8
Feb 21 04:05:21 witherspoon kernel[1038]: [127298.880000] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
Feb 21 04:05:21 witherspoon systemd[1]: Starting Deletes read-only and kernel ubi volume 9fc1c935...
Feb 21 04:05:22 witherspoon systemd[1]: Started Set U-Boot environment variable.
Feb 21 04:05:22 witherspoon obmc-flash-bios[23274]: libubi: error!: cannot read "/sys/class/ubi/ubi8_1/type"
Feb 21 04:05:22 witherspoon obmc-flash-bios[23274]:         error 19 (No such device)
Feb 21 04:05:23 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Feb 21 04:05:23 witherspoon obmc-flash-bios[23274]: ubinfo: error!: libubi failed to probe volume 1 on ubi8
Feb 21 04:05:23 witherspoon obmc-flash-bios[23274]:         error 19 (No such device)
Feb 21 04:05:23 witherspoon kernel: ubi8 error: ubi_open_volume: cannot open device 8, volume 0, error -16
Feb 21 04:05:23 witherspoon kernel[1038]: [127300.770000] ubi8 error: ubi_open_volume: cannot open device 8, volume 0, error -16
Feb 21 04:05:23 witherspoon obmc-flash-bios[23274]: ubirmvol: error!: cannot UBI remove volume
Feb 21 04:05:23 witherspoon obmc-flash-bios[23274]:           error 16 (Device or resource busy)
Feb 21 04:05:24 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Main process exited, code=dumped, status=11/SEGV
Feb 21 04:05:24 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Unit entered failed state.
Feb 21 04:05:24 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Failed with result 'core-dump'.
Feb 21 04:05:24 witherspoon systemd[1]: Started Clean up UBI volumes unattached to a Host version.
Feb 21 04:05:24 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Feb 21 04:05:24 witherspoon systemd[1]: Starting Updates the u-boot variable to point BMC version to 376586cb...
Feb 21 04:05:24 witherspoon systemd[1]: Started Remove UBIFS volume pnor-rw-fe429cef.
Feb 21 04:05:25 witherspoon kernel: block ubiblock4_0: released
Feb 21 04:05:25 witherspoon kernel[1038]: [127302.180000] block ubiblock4_0: released
Feb 21 04:05:25 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Service hold-off time over, scheduling restart.
```

software enumerate before DeleteAll 

```
$curl -c cjar -b cjar -k -X GET https://${BMC_IP}/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/376586cb": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Priority": 0, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "ibm-v2.1-105-g4397421-r1-0-gbeb0b18", 
      "associations": [
        [
          "inventory", 
          "activation", 
          "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
      ]
    }, 
    "/xyz/openbmc_project/software/376586cb/inventory": {
      "endpoints": [
        "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
      ]
    }, 
    "/xyz/openbmc_project/software/376586cb/software_version": {
      "endpoints": [
        "/xyz/openbmc_project/software"
      ]
    }, 
    "/xyz/openbmc_project/software/997aa69b": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
[... 114 lines cut ...]
        "/xyz/openbmc_project/inventory/system/chassis"
      ]
    }, 
    "/xyz/openbmc_project/software/997aa69b/software_version": {
      "endpoints": [
        "/xyz/openbmc_project/software"
      ]
    }, 
    "/xyz/openbmc_project/software/active": {
      "endpoints": [
        "/xyz/openbmc_project/software/376586cb", 
        "/xyz/openbmc_project/software/997aa69b"
      ]
    }, 
    "/xyz/openbmc_project/software/functional": {
      "endpoints": [
        "/xyz/openbmc_project/software/376586cb", 
        "/xyz/openbmc_project/software/997aa69b"
      ]
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

gtmills, 2018-02-28:
The delete appears to have worked but core dumped in the process.

rfrandse, 2018-03-01:
<https://gerrit.openbmc-project.xyz/9278> item_updater: Fix DeleteAll
Fixes  #2954 xyz.openbmc_project.Software.BMC.Updater.service core dump on DeleteAll.

closed 2018-03-01 by geissonator