openbmc/openbmc #2957: GUI: "Last power operation" time is incorrect

gtmills, 2018-03-01:
[image]

The "Last power operation" time is incorrect. The time shown is the last refresh time instead of the last time a power operation was performed.

https://github.com/openbmc/phosphor-webui/blob/master/app/server-control/controllers/power-operations-controller.html#L8

iftekharuli, 2018-04-10:
Is there a rest call we can make to get the time for the last power operations?

gtmills, 2018-07-25:
Code is here for the last power operation: https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-state-manager/+/11475/

rfrandse, 2018-08-03:
<https://gerrit.openbmc-project.xyz/11762> Last power operation
Resolves  #2957 GUI: "Last power operation" time is incorrect

closed 2018-08-11 by geissonator