openbmc/openbmc #2960: GUI: After uploading the firmware page should display the new uploaded image

gtmills, 2018-03-01:
When uploading an image the firmware page does not refresh and show the new image.
To get the new image to show, a user must refresh the page.
The upload image code needs to wait until the image is uploaded, untarred, and a new software D-Bus object is created and then refresh the page. Currently, it just waits until a good response code is returned then refreshes. This is too soon since the new object has not been created yet.  

https://github.com/openbmc/phosphor-webui/blob/master/app/configuration/controllers/firmware-controller.js#L89

gtmills, 2018-03-01:
https://github.com/openbmc/openbmc/issues/2939 is the story to return the id of the new D-Bus software object. The REST upload should return the id after untarring and creating the image.

rfrandse, 2018-04-16:
<https://gerrit.openbmc-project.xyz/10128> Display error if error on upload
Resolves  #2960 GUI: After uploading the firmware page should display the new uploaded image

closed 2018-04-16 by geissonator